serialport/node-serialport #1864: SerialPort with Pinpad Device

aco7342, 2019-05-08:
### Summary of Problem

I developed an EFT / TEF application on windows 10 using a USB pinpad, however the communication is based on the exchange of messages between my application and an application on the pinpad known as a shared library, which has the EMV commands for use in the card.

I made a small test program that sends and waits for the response to three commands: Closing, Opening and Discovering pinpad details, manufacturer, serial number, firmware version, etc.

but I have two approaches and two problems

to test scenario (1) set MODOFECHA=1, another scenario (2) set MODOFECHA = 0
(Please answer all 3)

- What are you trying to do?
the application starts by sending a logical close command with the shared library and waits for response, then sends the opening command with the shared library waiting for response and then the command to discover details of the pinpad is also waiting for response and will do this for 500 iterations

- What happens? 2 cenarios same source to test
1) when closing the serial port and cleaning the port controller (port = null)
I have problem opening the serial on the i5 / core client machine, but this does not happen on my i7 / core machine

2) when I close the serial port and do not clean the port controller, I have a problem with port.on (date) and with each port.write, the higher the number of shots occur

- What should have happened?
1) open succesfully
2) a single receipt shot 

### Code to Reproduce the Issue
see inside zip file

```

### Versions, Operating System and Hardware

6 May 17:45:40:522 - [info] Electron Version:2.0.9
6 May 17:45:40:538 - [info] Platform:win32
6 May 17:45:40:538 - [info] CPUs:Intel(R) Core(TM) i7-7500U CPU @ 2.70GHz - 2904Mhz

- SerialPort@?  7.1.5 extract from package.json
- Node.js v? -- v8.15.0
- Windows? Linux? Mac? **_win32_**    -- App(Win32)/(Machine)Windows10x64
- Hardware and chipset? (Prolific/FTDI/Other)
![pinpad](https://user-images.githubusercontent.com/8000302/57339901-ea3cac80-7109-11e9-8d91-8134f480e875.png)
[Suporte.zip](https://github.com/serialport/node-serialport/files/3154910/Suporte.zip)
[Biblioteca compartilhada - v108a.pdf](https://github.com/serialport/node-serialport/files/3154913/Biblioteca.compartilhada.-.v108a.pdf)

reconbot, 2019-05-08:
Can you reduce the problem down to a single file? A test case that does the minimum makes it much easier to debug.

Also it might help you to enable debug mode https://serialport.io/docs/en/guide-debugging

aco7342, 2019-05-09:
Solution: Destroying the Event Emitter Listener  [   port.removeListener('data',.....)   ],

However the error in opening the communication port the first time persists, after some error is opened

closed 2019-05-09 by aco7342