serialport/node-serialport #1866: Expose bcdDevice into Serialport.list

danni, 2019-05-09:
"bcdDevice is used to provide a device version number."

This can be used to show differences in protocol version for different CDC devices.

The value is in binary coded decimal with a format of 0xJJMN where JJ is the major version number, M is the minor version number and N is the sub minor version number. e.g. 2.0 is reported as 0x0200, 1.1 as 0x0110 and 1.0 as 0x0100.

reconbot, 2019-05-09:
Can you give more information, os, an example device etc

danni, 2019-05-09:
OS would be all OSes. It's a piece of the USB device info block. That is typically used to identify a product revision before you connect. We have a device that uses it to indicate what serial protocol to talk to it with.

reconbot, 2019-05-10:
Ahh... that's bit harder to get as we don't always use USB information. But I'll tag this as a feature request. I think some usb info might be useful.

danni, 2019-05-10:
It's available within the block that contains the vendor and product IDs. It's available in IOKit in the same struct. I suspect it's equally accessible in Windows.

Granjow, 2022-06-09:
I currently have a situation where the bcdDevice is the only way to distinguish two different devices. So having this property exposed would be useful.