serialport/node-serialport #1867: Can't use in Chrome ??

zhoubo2015, 2019-05-10:
### Summary of Problem

(Please answer all 3)

- What are you trying to do?
 I want read data from COM in Chrome
- What happens?
bindings.js?f3ff:178 Uncaught TypeError: Cannot read property 'indexOf' of undefined
    at Function.getFileName (bindings.js?f3ff:178)
    at bindings (bindings.js?f3ff:82)
    at eval (linux.js?a6c0:1)
    at Object../node_modules/@serialport/bindings/lib/linux.js (app.js:782)
    at __webpack_require__ (app.js:679)
    at fn (app.js:89)
    at Object.eval (index.js?498a:14)
    at eval (index.js:17)
    at Object../node_modules/@serialport/bindings/lib/index.js (app.js:768)
    at __webpack_require__ (app.js:679)
- What should have happened?

### Code to Reproduce the Issue

```js
    var SerialPort = require('serialport')
    var portName = 'COM3'
    var serialPort = new SerialPort(
      'COM3', {
        baudRate: 9600,
        dataBits: 8,
        parity: 'none',
        stopBits: 1,
        flowControl: false,
        autoOpen: false
      }, false)
    serialPort.open(function (error) {
      if (error) {
        console.log('open:' + portName + 'error:' + error)
      } else {
        console.log('is opened')
      }
    })

// Code
```

### Versions, Operating System and Hardware

- SerialPort@7.1.5
- Node.js v10.15.3
- Windows
- Hardware and chipset? (Prolific/FTDI/Other)

zhoubo2015, 2019-05-10:
[image]

zhoubo2015, 2019-05-10:
The front frame is vue

reconbot, 2019-05-10:
Serialport only works in electron or nodejs. It also doesn’t webpack well and needs to be excluded.

zhoubo2015, 2019-05-13:
> Serialport only works in electron or nodejs. It also doesn’t webpack well and needs to be excluded.

Ok,  I understand. Thank you.

closed 2019-05-13 by zhoubo2015