serialport/node-serialport #1868: Writing to COM port (GetOverlappedResult): The parameter is incorrect

JsseL, 2019-05-10:
### Summary of Problem

**What are you trying to do?**
Trying to write in the simplest manner to an open (emulated) COM port on Windows 10. I'm using serialport-repl to cut minimize the different factors.

**What happens?**
I get an error saying: `Writing to COM port (GetOverlappedResult): The parameter is incorrect`

**What should have happened?**
Write should have succeeded.

### Code to Reproduce the Issue

```cmd
C:\Users\Me\project>serialport-repl COM12
port = SerialPort("COM12", { autoOpen: false })
globals { SerialPort, portName, port }
> port.open()
undefined
> port.isOpen
true
> port.write('Yep', 'utf8', () => { console.log('write done'); });
true
> write done
(node:5300) UnhandledPromiseRejectionWarning: Error: Writing to COM port (GetOverlappedResult): The parameter is incorrect
(node:5300) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 1)
(node:5300) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
> port.isOpen
false
```

### Versions, Operating System and Hardware

- @serialport/repl@1.1.5
- serialport@7.1.5
- Node.js 8.11.1
- Windows 10
- Emulated COM port with ["Free" Virtual Serial Port](https://freevirtualserialports.com/)

## My question

How can I find out which "parameter is incorrect" as the error message states `The parameter is incorrect`? Or is there something else that could lead me to the right track?

reconbot, 2019-05-10:
Nobody has ever gotten free virtual serialport to work right with serialport. I’m more than happy to work with their support but I don’t think anyone has ever asked.

JsseL, 2019-05-13:
@reconbot Thanks. Better not to get stuck with that then. This is probably good information for others too. 👍

To have a chance of things going forward, I sent them information about this issue via their contact form. Let's see if they respond.

Thanks for the answer and thanks for the great work you do. 💯 You are empowering interesting projects! Will close the issue for now.

closed 2019-05-13 by JsseL

JsseL, 2019-05-15:
@reconbot I managed to contact the people behind Free Virtual Serial Port and was adviced to "bridge" two ports and use one of the bridged ports. Tried that and got past the `Writing to COM port (GetOverlappedResult): The parameter is incorrect` error.

I still have a problem with "draining". Serialport.io reports indefinitely that the port needs to wait for draining after the first message. I'll report more if I manage to get the flow control working.

Anyways, they seemed to be happy to help in case you need to contact them.

reconbot, 2019-05-16:
Would you be able to add a section to our docs on testing describing how to make it work? I think a lot of people would find it useful.

The docs have recently moved over here
https://github.com/serialport/website/blob/master/docs/guide-testing.md

JsseL, 2019-05-16:
@reconbot Got it actually working. Drain related problem was in my code. So the port bridging did eventually work nicely.

Made a [PR](https://github.com/serialport/website/pull/6) also as you suggested.