serialport/node-serialport #1869: TypeError: Error.captureStackTrace is not a function

MarceloSavian, 2019-05-13:
Hi,

I created a React app with the command Creat-React-App, and i have to use the serial port to print a something in a thermal print.

I already got some results, but executing the file with the command node app.js in the CMD.

I want to put this in a inteface, so i have to run the npm start, and by doing this i got the following error. I just import the serial port with const SerialPort = require('serialport');. Don't find anything related here, what i have to do ?

> TypeError: Error.captureStackTrace is not a function

> }; // run the 'prepareStackTrace' function above

> Error.captureStackTrace(dummy); //Error is here

>   dummy.stack; // cleanup

> Error.prepareStackTrace = origPST;

reconbot, 2019-05-13:
This is probably a webpack / electron issue.

MarceloSavian, 2019-05-14:
But do i have to install electron to use this ?

reconbot, 2019-05-14:
Sorry I assumed you were using electron, this doesn’t work in a browser as they don’t yet have serialports.

MarceloSavian, 2019-05-14:
Is the 10.15.3

MarceloSavian, 2019-05-14:
I changed the import to const SerialPort = require('@serialport/stream');, and now i am receiving a "No Binding set on `SerialPort.Binding`" as an error. Is this a problem with the node or serialport package version ?
Do i have to add any other package ?

reconbot, 2019-05-20:
You have to add a the bindings package. `@serialport/bindings`

closed 2019-07-26 by stale[bot]