serialport/node-serialport #1870: Darwin bindings aren't properly failling when opening a locked port

reconbot, 2019-05-16:
## 🐞Bug

From https://github.com/serialport/node-serialport/issues/1826#issuecomment-487119926 we have a bug where darwin bindings aren't properly failing when opening a locked port

>❯ DEBUG=serialport/binding* yarn start
> yarn run v1.13.0
> $ node src/index.js
> serialport/bindings loading DarwinBinding +0ms
> serialport/binding-abstract open +0ms
> serialport/binding-abstract open +3ms
> Error: Error Resource temporarily unavailable Cannot lock port
> serialport/bindings/poller Creating poller +0ms
> Connection with undefined opened.
> serialport/binding-abstract read +4ms

This should case `open` to reject.

eternalharvest, 2020-05-14:
@reconbot 
Thank u for developping such a nice library!
I was very impressed with this library because it makes manipulating asynchronous serial communication devices very easy.

But I have same issue on unix (Docker on Linux) bindings aren't properly failing.
I think my issue is related to this.

I expect the port trigger error when opening the port locked.
But port can be opened without error, and i can call `write()` method of the port instance.
Though i can't get response from the port.

Sometimes, when I open the port already locked (I'm not sure if the port is actually locked by OS), pre-existing stream seems to be broken.
I can't read any response anymore on the port opened first.
I'm not debugged this issue yet, but i think this is a bug.
Should I create new issue for this?

#### Note
* Node Version: v14.2.0
* serialport version: 9.0.0
* I'm using flow control `RTSCTS`

##### My Configurations
```javascript
const port = new SerialPort('/dev/ttyUSB0', {
  baudRate: 9600,
  dataBits: 7,
  stopBits: 1,
  parity: 'even',
  rtscts: true,
  vmin: 0,
  vtime: 10,
  lock: true,
});
```

Best regards.

eternalharvest, 2020-05-14:
@reconbot

Sorry, it seems the issue is related to docker's behaviour.
I'm tested this simple code below in docker container which is mounting `/dev/ttyUSB0`

```c:porttest.c
#include <sys/file.h>
#include <stdio.h>
#include <stdlib.h>
#include <unistd.h>
#include <errno.h>
#include <string.h>

int main(int argc, char **argv)
{
  int flags = (O_RDWR | O_NOCTTY | O_NONBLOCK | O_CLOEXEC | O_SYNC);
  int fd = open(argv[1], flags);

  if (-1 == fd) {
    fprintf(stderr, "Error %s Cannot lock port\n", strerror(errno));
    return EXIT_FAILURE;
  }

  if (-1 == flock(fd, LOCK_EX | LOCK_NB)) {
    fprintf(stderr, "Error %s Cannot lock port\n", strerror(errno));
    return EXIT_FAILURE;
  }

  fprintf(stderr, "Press any key to continue\n");
  getchar();
  if (-1 == close(fd)) {
    fprintf(stderr, "Error %s Cannot close port\n", strerror(errno));
    return EXIT_FAILURE;
  }

  return EXIT_SUCCESS;
}
```

When I run this code on docker container, `flock()` seems always success even if another docker container is locking / host is locking the port.
But when I enter 
 into the same container and run this test command, the error message "Error Resource temporarily unavailable Cannot lock port" is appeared on the console.
So, it seems the namespace of `flock()` is separated by docker container's boundary.

I apologize to you for a fuss over