serialport/node-serialport #1874: Serialport v7.x cannot install with electron v3.x

GongT, 2019-05-20:
### Summary of Problem

- What are you trying to do?
> yarn add serialport    
> ./electron -i -r serialport

- What happens?
Electron 3.x will crash:
```
/xxxx/electron -i: symbol lookup error: /XXXXX/node_modules/@serialport/bindings/build/Release/bindings.node: undefined symbol: _ZN2v816FunctionTemplate3NewEPNS_7IsolateEPFvRKNS_20FunctionCallbackInfoINS_5ValueEEEENS_5LocalIS4_EENSA_INS_9SignatureEEEiNS_19ConstructorBehaviorENS_14SideEffectTypeE
```
Use with electron v4.0.0 is ok.    
Manual rebuild with `node-gyp` will work on electron v3.    
I think it always download binary for electron v4.

### Versions, Operating System and Hardware

- SerialPort@7.1.5
- Electron v3.1.8
- Linux

reconbot, 2019-05-20:
I need more information or example code to give advise. I'm able to both download and build serialport for electron. I've started this repo to give an example of electron working correctly. https://github.com/serialport/electron-serialport-quick-start

GongT, 2019-05-21:
**It works very well on electron 4/5.**

Step to reproduce:
```bash 
mkdir /tmp/xxxxx && cd /tmp/xxxxx

wget https://github.com/electron/electron/releases/download/v3.1.9/electron-v3.1.9-linux-x64.zip
unzip electron-v3.1.9-linux-x64.zip

export npm_config_target=3.1.9
export npm_config_arch=x64
export npm_config_target_arch=x64
export npm_config_disturl=https://atom.io/download/electron
export npm_config_runtime=electron

npm install serialport

./electron -r serialport -i
```

> /tmp/xxxxx/electron -r -i serialport: symbol lookup error: /tmp/etest/node_modules/@serialport/bindings/build/Release/bindings.node: undefined symbol: _ZN2v816FunctionTemplate3NewEPNS_7IsolateEPFvRKNS_20FunctionCallbackInfoINS_5ValueEEEENS_5LocalIS4_EENSA_INS_9SignatureEEEiNS_19ConstructorBehaviorENS_14SideEffectTypeE

GongT, 2019-05-21:
```bash
cd node_modules/@serialport/bindings
node-gyp rebuild
cd -
./electron -r serialport -i
# works
```

```bash
cd node_modules/@serialport/bindings
../../.bin/prebuild-install --tag-prefix @serialport/bindings@ --verbose
cd -
./electron -r serialport -i
# does not work
```

prebuild-install output:
```
prebuild-install info begin Prebuild-install version 5.3.0
prebuild-install info looking for cached prebuild @ /home/GongT/.npm/_prebuilds/d52d5f-bindings-v2.0.8-electron-v64-linux-x64.tar.gz
prebuild-install info found cached prebuild
prebuild-install info unpacking @ /home/GongT/.npm/_prebuilds/d52d5f-bindings-v2.0.8-electron-v64-linux-x64.tar.gz
prebuild-install info unpack resolved to /tmp/etest/node_modules/@serialport/bindings/build/Release/bindings.node
prebuild-install info install Successfully installed prebuilt binary!
```

GongT, 2019-07-22:
/ping

closed 2019-09-27 by stale[bot]