serialport/node-serialport #1875: Serialport.list no response

popjxc, 2019-05-21:
### Summary of Problem

(Please answer all 3)

- What are you trying to do?
   Using serialport on electron app.
- What happens?
  No response after call Serialport.list
- What should have happened?
 return all my serial ports like serialport-list.
I try run the command line tool in the doc, it works well:

$ npx @serialport/list
/dev/tty.Bluetooth-Incoming-Port		
/dev/tty.usbserial-1420		FTDI
/dev/tty.usbserial-1410		FTDI

### Code to Reproduce the Issue
In the electron's main.js
```js
let win;

const SerialPort = require('serialport');
SerialPort.list().then(
  ports => ports.forEach(console.log),
  err => console.error(err)
);
console.log("SerialPort.list");

// Code
```
**All log in the console:**
> electron .

SerialPort.list
/dist/index.html
/dist/index.bundle.js
/dist/index.html
[5146:0521/141936.015835:ERROR:CONSOLE(24)] "Empty response arrived for script 'chrome-devtools://devtools/remote/serve_file/@033b679d965b0b3a085b6b911f9f8e216b8fbe32/cm_modes/cm_modes_module.js'", source: chrome-devtools://devtools/bundled/shell.js (24)
[5146:0521/141936.094472:ERROR:CONSOLE(108)] "Uncaught (in promise) Error: Could not instantiate: CmModes.DefaultCodeMirrorMimeMode", source: chrome-devtools://devtools/bundled/shell.js (108)
[5146:0521/141936.094744:ERROR:CONSOLE(108)] "Uncaught (in promise) Error: Could not instantiate: CmModes.DefaultCodeMirrorMimeMode", source: chrome-devtools://devtools/bundled/shell.js (108)

### Versions, Operating System and Hardware

- SerialPort@7.1.5?
- Node.js v? I use electron@5.0.1 and electron-rebuild
- Windows? Linux? Mac? Mac High Sierra
- Hardware and chipset? (Prolific/FTDI/Other)

popjxc, 2019-05-21:
I found another interesting thing, if SerialPort.list is called, the electron process can't quit when close the main window. I guess SerialPort.list missing resolve or reject the return promise, and make the process exit hang.

reconbot, 2019-05-21:
It’s the same code in the cli and the library. I would bet it’s an electron binary issue.

popjxc, 2019-05-22:
After switch from electron to nw.js, everything works well.  Thanks to @reconbot

closed 2019-05-22 by popjxc