serialport/node-serialport #1878: Johnny-Five v1.1.0 throws error with serialport in NW.js

djipco, 2019-05-24:
I have been using johnny-five v1.0.0 within NW.js projects for quite a while now. However, since version 1.1.0, my projects are giving me this error:

```Uncaught Missing serialport dependency```

If I switch back to Johnny-Five v1.0.0, everything works fine again. This happens on macOS and Windows 10. Johnny-Five now needs serialport v7.1.5 (this is the stated optional dependency version in `package.json`). I tried manually installing it  but it doesn't work either. 

I am not sure where the problem lies. I've filed [an issue](https://github.com/rwaldron/johnny-five/issues/1577) in the Johnny-Five repo as well. This issue also offers code to reproduce the problem. You do not even have to have an Arduino connected.

Ideas?

reconbot, 2019-05-24:
Answered over there. I don't really know how to support nw-js but I do have a lot of electron users who have great success.

djipco, 2019-07-10:
When you say "over there" where do you mean?

djipco, 2019-07-16:
For the sake of completeness, please note that this issue has been fixed in the Node.js code used by NW.js. This means that, starting with the [nightly build of July 17](https://dl.nwjs.io/live-build/nw39/07-17-2019/945a6ea62/v0.39.3/), this problem is resolved. 

Details are here: https://github.com/nwjs/node/commit/b2c926a38f1f73a199f90560f1e6e08cbfdd1aab

closed 2019-07-16 by djipco