serialport/node-serialport #1884: Prebuilt bindings (bindings-v2.0.8-electron-v70-win32-x64.tar.gz) contain v47 modules

jbollacke, 2019-06-04:
The prebuilt bindings provided in bindings-v2.0.8-electron-v70-win32-x64.tar.gz contain are module version v47 while they should be v70.

I have not checked other architectures (e.g. ia32) or targets.

josephlarralde, 2019-06-04:
I keep getting similar errors trying to use serialport with electron.
I've tried lots of electron and node versions combinations, rebuild seems to work fine each time, but when I run my electron app I keep getting these NODE_MODULE_VERSION errors ...
Any hints on the right way to integrate serialport in electron ?
BTW, I'm on osx mojave

jbollacke, 2019-06-04:
Well using electron v4.2.3 with serialport:7.1.5 works on my Windows 10 x64 machine at work.

Somehow it won't download the prebuilt bindings (its using a wrong URL) but I am tricking it by manually providing a cache in AppData\Roaming\...

bencefr, 2019-07-10:
Installing serialport for electron 5.0.6 also fails with NODE_MODULE_VERSION mismatch.

Josverl, 2019-08-30:
if only i had seen this before, i did not have to doubt myself as much

phhoef, 2019-09-15:
Same here for electron 5.0.10.
In addition I am getting also this error. It seems, that the prebuilt cannot be downloaded from Github.
I've downloaded it manually and put it in the cache folder. But feels a bit dirty ...

```
electron-builder  version=21.2.0
  • loaded configuration  file=package.json ("build" field)
  • rebuilding native dependencies  dependencies=@serialport/bindings@2.0.8 platform=darwin arch=x64
  • install prebuilt binary  name=@serialport/bindings version=2.0.8 platform=darwin arch=x64
  • build native dependency from sources  name=@serialport/bindings
                                          version=2.0.8
                                          platform=darwin
                                          arch=x64
                                          reason=prebuild-install failed with error (run with env DEBUG=electron-builder to get more information)
                                          error=prebuild-install info begin Prebuild-install version 5.3.2
    prebuild-install info looking for cached prebuild @ /Users/ph/.npm/_prebuilds/73d12e-bindings-v2.0.8-electron-v70-darwin-x64.tar.gz
    prebuild-install http request GET https://github.com/node-serialport/node-serialport/releases/download/v2.0.8/bindings-v2.0.8-electron-v70-darwin-x64.tar.gz
    prebuild-install http 404 https://github.com/node-serialport/node-serialport/releases/download/v2.0.8/bindings-v2.0.8-electron-v70-darwin-x64.tar.gz
    prebuild-install WARN install No prebuilt binaries found (target=5.0.9 runtime=electron arch=x64 libc= platform=darwin)
```

reconbot, 2019-10-06:
I've removed the bad build please follow #1920 for testing on the new versions.

closed 2019-10-06 by reconbot