serialport/node-serialport #1891: Different Node.js version

phhoef, 2019-06-19:
### Summary of Problem
When using Electron 5 with Serialport 7.1.5 I am getting the following error.
I have a `postinstall` task that runs `electron-builder install-app-deps`.
As far as I know, this should re-compile the native dependencies against the appropriate Electron version.
When using Electron 4, it's working without any issues.

```
App threw an error during load
Error: The module '/Users/Client/node_modules/@serialport/bindings/build/Release/bindings.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 47. This version of Node.js requires
NODE_MODULE_VERSION 70. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).
    at process.func [as dlopen] (electron/js2c/asar.js:155:31)
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:850:18)
    at Object.func [as .node] (electron/js2c/asar.js:155:31)
    at Module.load (internal/modules/cjs/loader.js:677:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:609:12)
    at Function.Module._load (internal/modules/cjs/loader.js:601:3)
    at Module.require (internal/modules/cjs/loader.js:715:19)
    at require (internal/modules/cjs/helpers.js:14:16)
    at bindings (/Users/ph/Documents/Projekte/prosimpl/ALFcon/Client/node_modules/bindings/bindings.js:112:48)
    at Object.<anonymous> (/Users/ph/Documents/Projekte/prosimpl/ALFcon/Client/node_modules/@serialport/bindings/lib/darwin.js:1:36)
```

I am not sure, if this is a problem of Serialport or Electron or Electron-Bulider?

### Versions, Operating System and Hardware

- SerialPort@? v7.1.5
- Node.js v? v12.4.0
- Electron v5.0.4
- Mac

MedAmineBenyettou, 2019-06-19:
@phhoef  try this :

https://github.com/song940/node-escpos/issues/214#issuecomment-502806807

It worked for me, but gave me a strange behavior after.

phhoef, 2019-06-19:
Thanks for your reply - but your answer also confuses me. 
Ad far as I know, running electron-builder install-app-depsˋ should completely replace the ˋelectron-rebuildˋ. 

Anyhow, I tried running ˋelectron-rebuildˋ, but with no success.

sontek, 2019-06-21:
I'm experiencing the same issue.  Did older versions work?  I'm actually doing both because I didn't know what was necessary:

```
    "postinstall": "yarn rebuild && electron-builder install-app-deps",
    "rebuild": "electron-rebuild -f -o node-hid && electron-rebuild -f -o serialport",
```

phhoef, 2019-06-21:
Yes, my configuration is working, when using electron 4. 
as mentioned, I have a postinstall task for rebuilding the native dependencies: `electron-builder install-app-deps`

I‘ll try your `rebuild` task, maybe this is working for me ...

sontek, 2019-06-21:
The rebuild worked for me using `node-hid` but doesn't seem to work for this library:

[image]

phhoef, 2019-06-22:
Same here, rebuild does not work for `serialport`. :-(

3478349671, 2019-06-25:
I also encountered the same problem, using the latest version of electronic (5.0.4) node (12.4.0), serialport (7.1.5), and electronic back to 4.2.4.

Kosik-Chris, 2019-06-25:
MedAmineBenyettou comment solution worked for me. Rebuild worked for Zigbee.

sanzog03, 2019-06-27:
`./node_modules/.bin/electron-rebuild` appended at the run script worked for me.

haozewu, 2019-07-10:
I also encountered this problem, it still doesn't work in all solution above.

yikuo123, 2019-07-29:
the same problem

huoyunjie, 2019-08-09:
I also encountered the similar issue when I want to create a vscode plugin with serialport. 
My env:
windows 10 x64
serialport: 1.37.0
nodejs: 12.7.0
Activating extension `undefined_publisher.helloworld` failed:  The module '\\?\f:\tool_developer\vscode\helloworld\node_modules\@serialport\bindings\build\Release\bindings.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 72. This version of Node.js requires
NODE_MODULE_VERSION 69. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).

If I change the nodejs version to 10.11.0, also has this issue. But the error information is different:
Activating extension `undefined_publisher.helloworld` failed:  The module '\\?\f:\tool_developer\vscode\helloworld\node_modules\@serialport\bindings\build\Release\bindings.node'
was compiled against a different Node.js version using
**NODE_MODULE_VERSION 64**. This version of Node.js requires
**NODE_MODULE_VERSION 69**. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).

huoyunjie, 2019-08-31:
Hi guys, I have fixed this issue: use serialport in vscode plugin.

steps:
1. install electron-rebuild
npm install --save-dev electron-rebuild

2. install serialport
npm install serialport

3. rebuild
./node_modules/.bin/electron-rebuild --version 4.2.7
NOTE: 4.2.7 is electron version, perhaps you need to change for different vscode version.

closed 2019-11-06 by stale[bot]