serialport/node-serialport #1894: readline parser problem when stream consists of mix of ASCII and raw data

HemantJoshi, 2019-06-29:
### Summary of Problem
readline parser doesn't parse correctly, when the stream consists of mix of ASCII and raw data
(Please answer all 3)

- What are you trying to do?
Using readline parser to parse a stream which is mix of ASCII and raw data.
- What happens?
The raw data is not parsed correctly. See the code below.
The stream(line) starts with ASCII 'Hi there!' followed by raw byte (0xFA) followed by '\r\n'
- What should have happened?
The last byte should be (0xFA). But got (0xFD).
However when the higher nibble is 0 the data is parsed correctly. I didn't check for all values.

### Code to Reproduce the Issue

```js
const SerialPort = require('serialport')

// Code
```
const Readline = require('@serialport/parser-readline')
const port = new SerialPort('/dev/ttyUSB0')

const parser = port.pipe(new Readline({ delimiter: '\r\n', includeDelimiter: false }))
parser.on('data', console.log)

var write_buffer = Buffer.alloc(1);

port.write('Hi there!');
write_buffer[0] = 0xFA;
port.write(write_buffer);
port.write('\r\n');

parser.on('data', (data) => {

var read_buffer = Buffer.from(data, 'binary');
console.log(read_buffer.toString('hex'));
});

### Versions, Operating System and Hardware

- SerialPort@? v7.1.5
- Node.js v? v10.16.0
- Windows? Linux? Mac? Linux
- Hardware and chipset? (Prolific/FTDI/Other) CP2102

reconbot, 2019-06-29:
Looks like a good test case, however it is not safe to reuse the buffer.
-- 
---
Francis Gulotta
wizard@roborooter.com

HemantJoshi, 2019-08-01:
Since I realised that I need a custom parser, modified @serialport/parser-delimiter. It is working fine.
Also I think that there is no point in generalising and expecting the custom parser into serialport release.
So should I keep the issue as is or close it?

reconbot, 2019-08-20:
We can close it.

closed 2019-08-20 by reconbot