serialport/node-serialport #1895: Serial port connect electron

byarchil, 2019-07-02:
- Trying to read pin with johnny-five.
-  Error :NODE_MODULE_VERSION 64. This version of Node.js requires NODE_MODULE_VERSION 70.      Please try re-compiling or re-installing
- I do not know work on an empty project

### Code to Reproduce the Issue

```js
const serialport = new SerialPort(path)

(node:5746) UnhandledPromiseRejectionWarning: Error: The module '/Users/test/client/node_modules/@serialport/bindings/build/Release/bindings.node'
  was compiled against a different Node.js version using
  NODE_MODULE_VERSION 64. This version of Node.js requires
  NODE_MODULE_VERSION 70. Please try re-compiling or re-installing
  the module (for instance, using `npm rebuild` or `npm install`).
      at process.func [as dlopen] (electron/js2c/asar.js:155:31)
      at Object.Module._extensions..node (internal/modules/cjs/loader.js:857:18)
      at Object.func [as .node] (electron/js2c/asar.js:155:31)
      at Module.load (internal/modules/cjs/loader.js:677:32)
      at tryModuleLoad (internal/modules/cjs/loader.js:609:12)
      at Function.Module._load (internal/modules/cjs/loader.js:601:3)
      at Module.require (internal/modules/cjs/loader.js:715:19)
      at require (internal/modules/cjs/helpers.js:14:16)
      at bindings (/Users/intrepid/Documents/Project/DUYURUBU_CLIENT/db-client/node_modules/bindings/bindings.js:112:48)
      at Object.<anonymous> (/Users/intrepid/Documents/Project/DUYURUBU_CLIENT/db-client/node_modules/@serialport/bindings/lib/darwin.js:1:36)
      at Module._compile (internal/modules/cjs/loader.js:815:30)
      at Object.Module._extensions..js (internal/modules/cjs/loader.js:827:10)
      at Module.load (internal/modules/cjs/loader.js:677:32)
      at tryModuleLoad (internal/modules/cjs/loader.js:609:12)
      at Function.Module._load (internal/modules/cjs/loader.js:601:3)
      at Module.require (internal/modules/cjs/loader.js:715:19)

// Code
```

**My Test :** 

**npm rebuild** no problem

./node_modules/.bin/electron-rebuild  it gives an error

```js
An unhandled error occurred inside electron-rebuild
gyp info it worked if it ends with ok
gyp info using node-gyp@4.0.0
gyp info using node@10.16.0 | darwin | x64
gyp info spawn /usr/bin/python
gyp info spawn args [ '/Users/demo/Documents/Project/DD/db-client/node_modules/electron-rebuild/node_modules/node-gyp/gyp/gyp_main.py',
gyp info spawn args   'binding.gyp',
gyp info spawn args   '-f',
gyp info spawn args   'make',
gyp info spawn args   '-I',
gyp info spawn args   '/Users/demo/Documents/Project/DD/db-client/node_modules/avrgirl-arduino/node_modules/serialport/build/config.gypi',
gyp info spawn args   '-I',
gyp info spawn args   '/Users/demo/Documents/Project/DD/db-client/node_modules/electron-rebuild/node_modules/node-gyp/addon.gypi',
gyp info spawn args   '-I',
gyp info spawn args   '/Users/demo/.electron-gyp/5.0.6/include/node/common.gypi',
gyp info spawn args   '-Dlibrary=shared_library',
gyp info spawn args   '-Dvisibility=default',
gyp info spawn args   '-Dnode_root_dir=/Users/demo/.electron-gyp/5.0.6',
gyp info spawn args   '-Dnode_gyp_dir=/Users/demo/Documents/Project/DD/db-client/node_modules/electron-rebuild/node_modules/node-gyp',
gyp info spawn args   '-Dnode_lib_file=/Users/demo/.electron-gyp/5.0.6/<(target_arch)/node.lib',
gyp info spawn args   '-Dmodule_root_dir=/Users/demo/Documents/Project/DD/db-client/node_modules/avrgirl-arduino/node_modules/serialport',
gyp info spawn args   '-Dnode_engine=v8',
gyp info spawn args   '--depth=.',
gyp info spawn args   '--no-parallel',
gyp info spawn args   '--generator-output',
gyp info spawn args   'build',
gyp info spawn args   '-Goutput_dir=.' ]
gyp info spawn make
gyp info spawn args [ 'BUILDTYPE=Release', '-C', 'build' ]
[... 390 lines cut ...]
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:248:12)
gyp ERR! System Darwin 18.6.0
gyp ERR! command "/usr/local/bin/node" "/Users/demo/Documents/Project/DD/db-client/node_modules/electron-rebuild/node_modules/.bin/node-gyp" "rebuild" "--target=5.0.6" "--arch=x64" "--dist-url=https://electronjs.org/headers" "--build-from-source"
gyp ERR! cwd /Users/demo/Documents/Project/DD/db-client/node_modules/avrgirl-arduino/node_modules/serialport
gyp ERR! node -v v10.16.0
gyp ERR! node-gyp -v v4.0.0
gyp ERR! not ok

Failed with exit code: 1
    at SafeSubscriber._error (/Users/demo/Documents/Project/DD/db-client/node_modules/spawn-rx/lib/src/index.js:267:84)
    at SafeSubscriber.__tryOrUnsub (/Users/demo/Documents/Project/DD/db-client/node_modules/rxjs/internal/Subscriber.js:205:16)
    at SafeSubscriber.error (/Users/demo/Documents/Project/DD/db-client/node_modules/rxjs/internal/Subscriber.js:156:26)
    at Subscriber._error (/Users/demo/Documents/Project/DD/db-client/node_modules/rxjs/internal/Subscriber.js:92:26)
    at Subscriber.error (/Users/demo/Documents/Project/DD/db-client/node_modules/rxjs/internal/Subscriber.js:72:18)
    at MapSubscriber.Subscriber._error (/Users/demo/Documents/Project/DD/db-client/node_modules/rxjs/internal/Subscriber.js:92:26)
    at MapSubscriber.Subscriber.error (/Users/demo/Documents/Project/DD/db-client/node_modules/rxjs/internal/Subscriber.js:72:18)
    at SafeSubscriber._next (/Users/demo/Documents/Project/DD/db-client/node_modules/spawn-rx/lib/src/index.js:242:65)
    at SafeSubscriber.__tryOrUnsub (/Users/demo/Documents/Project/DD/db-client/node_modules/rxjs/internal/Subscriber.js:205:16)
    at SafeSubscriber.next (/Users/demo/Docume
### Versions, Operating System and Hardware

- "serialport": "^7.1.5",
-  "electron": "^5.0.4",
- Node.js v10.16.0
- Mac Mojave 10.14.5

intrepid-tr, 2019-07-03:
It's troublesome. but I can recommend this to you
**sudo rm -rf  /node_modules**
create binding.gyp inside the project and get the contents like this
`{
  "targets": [
    {
      "target_name": "binding",
       'sources' : [  "<!@(node -p \"require('fs').readdirSync('./src').map(f=>'src/'+f).join(' ')\")" ],
    }
  ]
}
`
node-gyp configure Check the correctness of your configuration.

**run terminal :** open  /Users/demo/.electron-gyp/5.0.6/include/node/v8.h
and fix 

current file
**`protected:**
  V8_INLINE HandleScope() {}

  void Initialize(Isolate* isolate);

  static internal::Object** CreateHandle(internal::Isolate* isolate,
                                         internal::Object* value);`

must be

`**public:**
  V8_INLINE HandleScope() {}

  void Initialize(Isolate* isolate);

  static internal::Object** CreateHandle(internal::Isolate* isolate,
                                         internal::Object* value);`

**In short you need to translate the protected part to public**
**and run agin** /node_modules/.bin/electron-rebuild

byarchil, 2019-07-03:
> It's troublesome. but I can recommend this to you
> 
> Create the **binding.gyp** file for the project. and add the following content
> 
> ```
>   "targets": [
>     {
>       "target_name": "binding",
>        'sources' : [  "<!@(node -p \"require('fs').readdirSync('./src').map(f=>'src/'+f).join(' ')\")" ],
>     }
>   ]
> }
> ```
> 
> node-gyp configure Check the correctness of your configuration.
> 
> **run terminal :** open /Users/demo/.electron-gyp/5.0.6/include/node/v8.h
> and fix
> 
> **Current File**
> 
> ```
> protected:
>   V8_INLINE HandleScope() {}
> 
>   void Initialize(Isolate* isolate);
> 
>   static internal::Object** CreateHandle(internal::Isolate* isolate,
>                                          internal::Object* value);`
> ```
> 
> **Must Be**
> 
> ```
> public:
>   V8_INLINE HandleScope() {}
> 
>   void Initialize(Isolate* isolate);
> 
>   static internal::Object** CreateHandle(internal::Isolate* isolate,
>                                          internal::Object* value);
> ```
> 
> **In short you need to translate the protected part to public**
> **and run agin** ./node_modules/.bin/electron-rebuild

 this worked well for me, many thanks.

raha1923, 2019-07-27:
It worked for me just in dev env and get the same error after running the prod exported version?

```
serialport: v7.1.5
node: v8.9.4
electron: v5.0.2
electron-reload: v1.4.0
```

*scripts:*
```
"electron:mac": "npm run build:prod && electron-builder build --mac",
"electron:linux": "npm run build:prod && electron-builder build --linux"
```

closed 2019-10-10 by stale[bot]