serialport/node-serialport #1896: only 4096 data

fenggu, 2019-07-03:
### Summary of Problem

- What are you trying to do?
I want to receive the serial port data sent by the printer. He is a compressed package.
- What happens?
But I found that I can only receive 4096 size buffers no matter how much data I send.
- What should have happened?
How can I receive all the data?
### Code to Reproduce the Issue

```js
  port = new SerialPort(
    setting.port || 'COM1', {
      baudRate: Number(setting.baudRate) || 9600,
      dataBits: Number(setting.dataBits) || 8, 
      parity: setting.parity || 'none',
      stopBits: Number(setting.stopBits) || 1,
      hupcl: true,
      buffersize: 500,
      rtscts: true,
   }, false)
  port.on('open', function () { 
    log.info(setting.port + ' is opening')
    port.on('data', (data) => {
      console.log(data.length)
      // 4096
      if (!startFlag) {
        startFlag = true
        startRecord(setting, res)
      }
    })
  });

// Code
```

### Versions, Operating System and Hardware

- SerialPort@?
- Node.js v? 8.3.0
- Windows? Linux? Mac? Win10
- Hardware and chipset? (Prolific/FTDI/Other) Hardware

fenggu, 2019-07-03:
sometimes it`s 4097

reconbot, 2019-07-03:
If you need more data at once you’ll need to use a parser https://serialport.io/docs/guide-about#parsers

closed 2019-09-02 by fenggu