serialport/node-serialport #1897: Close Port after readline parser ends

pacmac, 2019-07-03:
Just started using serial port, thanks it's exactly what I need.

I have googled this and looked through the examples and documents, but I cannot seem to find how to detect the end of the readline event so that I can close the port.

The code executes, and I get the returned data, but after receiving I want to close the port and not use a persistent connection.

```
const cl = console.log;
const SerialPort = require('serialport')
const Readline = require('@serialport/parser-readline');
const port = new SerialPort('/dev/tty.SLAB_USBtoUART', { baudRate: 115200 });
const parser = new Readline();
//const parser = port.pipe(new Readline({ delimiter: '\r\n' }))
port.pipe(parser);
parser.on('data', function(line){
  console.log(`> ${line}`);
})

parser.on('finish', function(){
  port.close();
  cl('closed');
})
port.write("something \r\n");
```

What am I doing wrong ??

reconbot, 2019-07-03:
Sounds like you want to close it after the first data event?

pacmac, 2019-07-03:
Thanks, I tried that before I posted:

```
parser.on('data', function(line){
  console.log(`> ${line}`);
  port.close();
})
```

But then I don't receive the line data and the port appears to be closed before the line is outputted to the console.

reconbot, 2019-07-03:
A data event is a line of data, do you need two lines of data or something?

pacmac, 2019-07-04:
OK, just figured out that the device (REPL) is echoing the command that is sent to it which in this case is `"stamps\r"`, and this behavior cannot be switched off.

This is the reply:

```
> stamps
> [265181, 265181, 265181, 265182, 265182, 265288]
```

Is there a recommended way of handling this, I guess in one respect the echo is useful as it confirms that the device received the request, but this is resulting in 2 lines of data being sent.

reconbot, 2019-07-04:
I would just check the reply to see if it's the message you would like and then close the port if it is.

closed 2019-09-09 by stale[bot]