serialport/node-serialport #1898: Prebuild-install paths are broken, possibly due to repro move

Josverl, 2019-07-05:
### Summary of Problem

(Please answer all 3)

- What are you trying to do?
In trying to use prebuild-install to retrieve the bindings for multiple electron versions & platforms
`C:\develop\<project>\node_modules\@serialport\bindings>.\node_modules\.bin\prebuild-install.cmd --runtime electron --target 4.2.5 --arch x64 --platform win32 --verbose`

- What happens?
prebuild-install is not able to find 
```
prebuild-install info begin Prebuild-install version 5.3.0
prebuild-install info looking for cached prebuild @ C:\Users\josverl\AppData\Roaming\npm-cache\_prebuilds\e9e8d7-bindings-v2.0.8-electron-v69-win32-x64.tar.gz
prebuild-install http request GET https://github.com/node-serialport/node-serialport/releases/download/v2.0.8/bindings-v2.0.8-electron-v69-win32-x64.tar.gz
prebuild-install http 404 https://github.com/node-serialport/node-serialport/releases/download/v2.0.8/bindings-v2.0.8-electron-v69-win32-x64.tar.gz
prebuild-install WARN install No prebuilt binaries found (target=4.2.5 runtime=electron arch=x64 libc= platform=win32)
```

the same happens also when using the default platform, version & architecture 
```
C:\develop\pymakr-vsc\node_modules\@serialport\bindings>.\node_modules\.bin\prebuild-install.cmd --verbose
prebuild-install info begin Prebuild-install version 5.3.0
prebuild-install info looking for cached prebuild @ C:\Users\josverl\AppData\Roaming\npm-cache\_prebuilds\89c4e1-bindings-v2.0.8-node-v64-win32-x64.tar.gz
prebuild-install http request GET https://github.com/node-serialport/node-serialport/releases/download/v2.0.8/bindings-v2.0.8-node-v64-win32-x64.tar.gz
prebuild-install http 404 https://github.com/node-serialport/node-serialport/releases/download/v2.0.8/bindings-v2.0.8-node-v64-win32-x64.tar.gz
prebuild-install WARN install No prebuilt binaries found (target=10.15.1 runtime=node arch=x64 libc= platform=win32)
```
this makes it very hard and troublesome to build and distribute multi-platform packages 
also due to quite sparse documentation it is hard to find out why thinks are not working.

- What should have happened?

The prebuilds should be found , downloaded, and subsequently installed

it would be even better to have documentation on how to install / download for multiple platforms ( i.e. prepackage for 3 platforms for electron 4.2.5.

### Code to Reproduce the Issue
see above ..

### possible cause 

Comparing the URLs used by prebuild-install and the URLs currently used there appreatrs to be 2 mismatches : 

| what | link 
|------|--------
| prebuild-install | https://github.com/node-serialport/node-serialport/releases/download/v2.0.8/bindings-v2.0.8-electron-v69-win32-x64.tar.gz
| releases | https://github.com/serialport/node-serialport/releases/download/%40serialport%2Fbindings%402.0.8/bindings-v2.0.8-electron-v69-win32-x64.tar.gz

I notice that in the path there are in 2 place in the path that are incorrect 

| where | prebuild-install | git hosted 
|--------|--------|------
| path root folder | node-serialport  | serialport 
| version folder | v2.0.8 |@serialport/bindings@2.0.8|

### Versions, Operating System and Hardware

- SerialPort@bindings@2.0.7 and 2.0.8 
- Node.js - v10.15.1
- Windows 10.0.18362.10000
- Hardware : n/a

Josverl, 2019-07-05:
explaining the problem made me look at options to change the version prefix 

it turns out there is an option in `prebuild-install` to do that :-) : --tag-prefix 
**--tag-prefix @serialport/bindings@**

```
C:\develop\<project>\node_modules\@serialport\bindings>node .\node_modules\prebuild-install\bin.js --runtime electron --target 4.2.5 --arch x64 --platform darwin --tag-prefix @serialport/bindings@ --verbose
```
This results in a successful install of the binding for electron-mac-x64  build on a windows machine 

```
prebuild-install info begin Prebuild-install version 5.3.0
prebuild-install info looking for cached prebuild @ C:\Users\josverl\AppData\Roaming\npm-cache\_prebuilds\6a8d11-bindings-v2.0.8-electron-v69-darwin-x64.tar.gz
prebuild-install http request GET https://github.com/node-serialport/node-serialport/releases/download/@serialport/bindings@2.0.8/bindings-v2.0.8-electron-v69-darwin-x64.tar.gz
prebuild-install http 200 https://github.com/node-serialport/node-serialport/releases/download/@serialport/bindings@2.0.8/bindings-v2.0.8-electron-v69-darwin-x64.tar.gz
prebuild-install info downloading to @ C:\Users\josverl\AppData\Roaming\npm-cache\_prebuilds\6a8d11-bindings-v2.0.8-electron-v69-darwin-x64.tar.gz.22648-63b6b0d443a6b.tmp
prebuild-install info renaming to @ C:\Users\josverl\AppData\Roaming\npm-cache\_prebuilds\6a8d11-bindings-v2.0.8-electron-v69-darwin-x64.tar.gz
prebuild-install info unpacking @ C:\Users\josverl\AppData\Roaming\npm-cache\_prebuilds\6a8d11-bindings-v2.0.8-electron-v69-darwin-x64.tar.gz
prebuild-install info unpack resolved to C:\develop\<project>\node_modules\@serialport\bindings\build\Release\bindings.node
prebuild-install info install Successfully installed prebuilt binary!
```
Note that there is still a path mismatch , but apparently this is resolved somehow so I won't worry about that anymore.

Still , given the time I have spend to figure this out, I think this calls for some additional documentation.

If there is another  or a simpler solution to package serialport in a single multi-platform module, while still using the prebuilts, please let me know.

sacha67, 2019-07-08:
Had the same problem when I upgraded serialport in my project. 
@Josverl  Thank you for the solution.

JuanM04, 2019-07-16:
So, how do I solve it? When I run `electron-build -w` on mac, prebuild tries to fetch and the build stops

Josverl, 2019-07-16:
> So, how do I solve it? When I run `electron-build -w` on mac, prebuild tries to fetch and the build stops

@JuanM04, I do not see how this is related to the issue I raised, and your description is quite sparse. So I cannot help you.

JuanM04, 2019-07-16:
Nevermind, I solved it. I just commented here because hours of trying to fix it brought me here. What happened is that electron-builder uses prebuild-install when building an Electron app. There is somewhere an issue related to this, and I'm also going to post my solution there.

I just created a `.prebuild-installrc` to do it automatically for me:
```json
"scripts": {
  "postinstall": "echo \"tag-prefix=@serialport/bindings@\" > ./node_modules/@serialport/bindings/.prebuild-installrc && electron-builder install-app-deps"
}
```

Josverl, 2019-07-26:
I have create a script to download bindings for multiple versions & platforms and store them in a simple file structure, so they can be used to load the module dynamically.

https://github.com/Josverl/pymakr-vsc/blob/fix/SerialMultiPlatform/scripts/mp-download.ps1

its in powershell as i find that simpler the write / debug , but I think it can be transposed to just any other scripting language

Josverl, 2019-08-28:
It should be noted that it appears that a number of the published builds are actually incorrect 
https://github.com/serialport/node-serialport/issues/1920

phhoef, 2019-09-15:
@Josverl I am using macOS, but I've the same issue.
Thanks for your effort for describing your solution.

Unfortunately, I do not completely understand, what you've done.
You ran this command `C:\develop\<project>\node_modules\@serialport\bindings>node .\node_modules\prebuild-install\bin.js --runtime electron --target 4.2.5 --arch x64 --platform darwin --tag-prefix @serialport/bindings@ --verbose` and the important point is the `--tag-prefix`-flag, right?
Do you run this after every `yarn` or `npm install`? Is it sufficient to run it once?
Did you find a way to integrate it into a script of the `package.json`?
To cut a long story short: Is there a way to properly integrate your solution in order to make it work every time without any additional command execution?

Thanks

closed 2019-11-21 by stale[bot]