serialport/node-serialport #1903: Newline mystery

adamreisnz, 2019-07-23:
### Summary of Problem

We're sending a JSON payload through the serial port to a board that processes this, and to separate payload packages we terminate each payload with a newline character.

This was working as expected correctly up until yesterday. As of yesterday I noticed that the board was throwing errors saying our JSON payload was invalid. I checked all the data we were sending, and nothing whatsoever had changed on our end. The payload looked good and sane. Moreover, the board was responding to the payload correctly. The JSON error was showing up after the message had been processed.

After some experimentation, we decided to remove the newline character from our payload. This solved the problem, the errors were no longer being reported and the payloads were still being processed correctly.

We then tried to replicate this on our test rig, but found that we couldn't. The newline character was in fact necessary, and if omitted it resulted in timeouts, as the board was waiting for more payload. When included, it wasn't triggering any errors and everything worked as expected, just as it had on the other rig which suddenly started to exhibit this problem.

We're using the exact same firmware on the board and the exact same software on the Pi. Both are using the latest versions of @serialport packages. To be honest we're a bit baffled by this, as we're not sure what would be causing the difference in behaviour.

Questions:
- Is there anything in SerialPort that automatically appends a newline character?
- Anything that could be different in the bindings if they were rebuilt that could cause this difference in behaviour?
- Any way we can reliably test this and debug it further?

### Code to Reproduce the Issue

Some code omitted for brevity.

```js
const SerialPort = require('@serialport/stream');
const Bindings = require('@serialport/bindings');
const Readline = require('@serialport/parser-readline');

this.connection = new SerialPort(port, {
      baudRate,
      autoOpen: false,
      binding: Bindings,
    });

const transform = new Readline({delimiter: '\n', encoding: 'utf8'});
    this.parser = this.connection.pipe(transform);
//...

this.connection.open(error => {});

//...

const json = JSON.stringify(data);
const buffer = Buffer.from(json +'\n', 'utf8');

this.connection.write(buffer, error => {});

```

### Versions, Operating System and Hardware

- SerialPort@?
- Node.js v?
- Windows? Linux? Mac?
- Hardware and chipset? (Prolific/FTDI/Other)

reconbot, 2019-07-23:
No new lines works? That’s not possible with that parser, maybe there are new lines you’re not aware of?

Are you cloning the rpis sd card or just installed your code on two different rpis? My hunch is an os level setting with regards to the serialport got changed on one of the machines. There are some settings that can change line breaks and other potentially harmful for communication but good for terminals modifications.

If you duplicate the sd card the possibility of this isn’t present.

Lastly what json error? A json parse error? Are you able to print the payload it couldn’t parse as part of the error? We could be barking up the wrong tree.

adamreisnz, 2019-07-23:
Hi @reconbot, thanks for looking into this.

> No new lines works? That’s not possible with that parser, maybe there are new lines you’re not aware of?

Doesn't the parser only kick in for incoming data, not outgoing?

I hacked the UnixWrite code to add another line of logging, to debug exactly what contents were being written to the serial port. I can confirm that it didn't log any newline at the end. The hex contents were `7b ..... 7d` which corresponds to the JSON curly brackets `{ ..... }`.

When we did add a newline, the hex data became `7b ..... 7d 0a`, with the newline present at the end of the buffer. So not sure if something else further down the line (in the bindings?) appends a newline or not.

> Are you cloning the rpis sd card or just installed your code on two different rpis? My hunch is an os level setting with regards to the serialport got changed on one of the machines. There are some [...]

Nope! In this case it was the exact same Pi that was first running in our office, and then it was sent to the customer who installed it on site, without making any alterations. The only change I can conceive of, is that some of the packages got updated, but other than that nothing has changed. That's why we're so baffled as well.

> If you duplicate the sd card the possibility of this isn’t present.

It was the exact same USB stick in the same Pi.

> Lastly what json error? A json parse error? Are you able to print the payload it couldn’t parse as part of the error? We could be barking up the wrong tree.

Regretfully no, the error is thrown by the board which is already installed, and we don't currently have access to it. We'd need to hook up a debugging port to the board to capture what JSON payload is failing exactly.

The parsing error results from the board receiving incorrect, non-null JSON. But it only happens when we append the newline to the payload. If we don't, it all works fine.

Even with appending the newline, the JSON payload is parsed correctly and acted upon, with the error appearing about half a second later. Almost as if the newline is introducing a second packet which the board tries to parse.

adamreisnz, 2019-09-22:
Still not resolved

closed 2019-11-28 by stale[bot]