serialport/node-serialport #1904: port.close error: "asyncClose is not defined"

belargo, 2019-07-26:
### **Summary of Problem**
While closing port using port.close() - error "asyncClose is not defined" is returned and port stays open.
What are possible reasons for this error?

- I'm trying to close previously opened port
- Error "asyncClose is not defined" is generated and port remains open
- port should be closed

### Versions, Operating System and Hardware

- SerialPort@8.0.0
- Electron 5.0.2 with NodeJS v. 12.0
- Windows 10 Pro

icebob, 2019-08-30:
Same issue here.
- Serialport@8.0.0
- Node.js v10
- Windows 10

reconbot, 2019-09-18:
I've just published serialport@8.0.1 give it a try. We'll get 8 out of beta yet.

belargo, 2019-09-18:
Thanks. I'll try new version and let you know the results.

choppsta, 2019-09-25:
Issue is still present in 8.0.2.  Think it's because the following is missing from the top of bindings/lib/win32.js:
`const asyncClose = promisify(binding.close)`

PR #1946 (2019-09-26): Add missing asyncClose
Fix #1904
commit 4a3d2a01 (2019-09-26): fix: Add missing asyncClose (#1946)

Fix #1904
closed 2019-09-26 by reconbot