serialport/node-serialport #1905: PrimeOS build issue

apiel, 2019-07-31:
### Summary of Problem

- I am trying to install serialport on primeOS (android for x86)
- I am getting 12 warnings and 9 errors generated.
- I try to find a way to compile it without errors.

### Code to Reproduce the Issue

`npm install serialport --build-from-source`

output:
```

> serialport@6.2.2 install /data/data/com.termux/files/home/mhb-js/node_modules/zigbee-service/node_modules/serialport
> prebuild-install || node-gyp rebuild

make: Entering directory '/data/data/com.termux/files/home/mhb-js/node_modules/zigbee-service/node_modules/serialport/build'
  CXX(target) Release/obj.target/serialport/src/serialport.o
../src/serialport.cpp:41:39: error: no matching member function for call to 'ToString'
  v8::String::Utf8Value path(info[0]->ToString());
                             ~~~~~~~~~^~~~~~~~
/data/data/com.termux/files/home/.node-gyp/12.4.0/include/node/v8.h:2528:44: note: candidate function not viable: requires single argument 'context',
      but no arguments were provided
  V8_WARN_UNUSED_RESULT MaybeLocal<String> ToString(
                                           ^
/data/data/com.termux/files/home/.node-gyp/12.4.0/include/node/v8.h:2544:35: note: candidate function not viable: requires single argument 'isolate',
      but no arguments were provided
                    Local<String> ToString(Isolate* isolate) const);
                                  ^
../src/serialport.cpp:48:44: error: no matching member function for call to 'ToObject'
  v8::Local<v8::Object> options = info[1]->ToObject();
                                  ~~~~~~~~~^~~~~~~~
/data/data/com.termux/files/home/.node-gyp/12.4.0/include/node/v8.h:2532:44: note: candidate function not viable: requires single argument 'context',
      but no arguments were provided
  V8_WARN_UNUSED_RESULT MaybeLocal<Object> ToObject(
                                           ^
/data/data/com.termux/files/home/.node-gyp/12.4.0/include/node/v8.h:2546:35: note: candidate function not viable: requires single argument 'isolate',
      but no arguments were provided
                    Local<Object> ToObject(Isolate* isolate) const);
                                  ^
../src/serialport.cpp:95:18: warning: 'Call' is deprecated [-Wdeprecated-declarations]
  data->callback.Call(2, argv);
                 ^
../../../../nan/nan.h:1739:3: note: 'Call' has been explicitly marked deprecated here
  NAN_DEPRECATED inline v8::Local<v8::Value>
  ^
../../../../nan/nan.h:104:40: note: expanded from macro 'NAN_DEPRECATED'
# define NAN_DEPRECATED __attribute__((deprecated))
                                       ^
../src/serialport.cpp:113:44: error: no matching member function for call to 'ToObject'
  v8::Local<v8::Object> options = info[1]->ToObject();
                                  ~~~~~~~~~^~~~~~~~
/data/data/com.termux/files/home/.node-gyp/12.4.0/include/node/v8.h:2532:44: note: candidate function not viable: requires single argument 'context',
      but no arguments were provided
  V8_WARN_UNUSED_RESULT MaybeLocal<Object> ToObject(
                                           ^
/data/data/com.termux/files/home/.node-gyp/12.4.0/include/node/v8.h:2546:35: note: candidate function not viable: requires single argument 'isolate',
      but no arguments were provided
                    Local<Object> ToObject(Isolate* isolate) const);
                                  ^
../src/serialport.cpp:150:18: warning: 'Call' is deprecated [-Wdeprecated-declarations]
  data->callback.Call(1, argv);
                 ^
../../../../nan/nan.h:1739:3: note: 'Call' has been explicitly marked deprecated here
  NAN_DEPRECATED inline v8::Local<v8::Value>
  ^
../../../../nan/nan.h:104:40: note: expanded from macro 'NAN_DEPRECATED'
# define NAN_DEPRECATED __attribute__((deprecated))
                                       ^
../src/serialport.cpp:188:18: warning: 'Call' is deprecated [-Wdeprecated-declarations]
[... 129 lines cut ...]
gyp ERR! not ok 
npm WARN retry-axios@0.3.2 requires a peer of axios@* but none is installed. You must install peer dependencies yourself.
npm WARN mhb-js@1.0.0 No description
npm WARN mhb-js@1.0.0 No repository field.

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! serialport@6.2.2 install: `prebuild-install || node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the serialport@6.2.2 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /data/data/com.termux/files/home/.npm/_logs/2019-07-31T19_52_06_346Z-debug.log

```
(I installed python2 and build-essential)

### Versions, Operating System and Hardware

- serialport@6.2.2
- Node.js v12.4.0
- Linux / Android x86 --> primeOS
- Beelink BT3 Pro Mini PC Intel Atom x5-Z8350 Processor

foxxyz, 2019-08-06:
Having the same issue on MacOS 10.14.6. Any ideas?

serialport@6.2.2
Node.js v12.7.0
3.1 Ghz Intel Core i7

HipsterBrown, 2019-08-12:
@foxxyz @apiel Can you try using an older version of Node.js to compile serialport? v10.16.2 is the latest LTS version. Node.js v12 has some breaking changes for native dependencies that we're still working through.

foxxyz, 2019-08-12:
Thanks @HipsterBrown! Good to know. 

Just tried with Node v10.16.2 (MacOS 10.14.6) and serialport@6.2.2 compiles fine.

Is there any issue or board that we can track for the ongoing status on Nodev12 support?

HipsterBrown, 2019-08-14:
@foxxyz Are you able to use the latest version of serialport? That compiles against Node v12 successfully.

foxxyz, 2019-08-14:
I would, but this is occurring via a dependency in [`dmx`](https://github.com/node-dmx/dmx) which is still [pinned to 6.0.4](https://github.com/node-dmx/dmx/blob/master/package.json).

I will open an issue with them to update also.

closed 2019-08-30 by reconbot