serialport/node-serialport #1906: Electron, VueJS -

Wolrab1, 2019-08-01:
### Summary of Problem

I have an electron application which uses VueJS. I'm trying to make use of serial port to connect to a usb RFID reader. I can get this working without vue so I assume it's due to webpack. 

- What are you trying to do?
Use SerialPort in my electron.js main file rather than my renderer. 

- What happens?
When defining SerialPort I get the following error:

```
App threw an error during load
TypeError: Path must be a string. Received undefined
```

- What should have happened?
The package should be available in my main.js file which is where I define my electron windows etc. 

### Code to Reproduce the Issue

**Main.js** 
```js

'use strict'

import { app, protocol, BrowserWindow, ipcMain, ipcRenderer, dialog, remote} from 'electron'
import * as path from 'path'
import { format as formatUrl } from 'url'
import { createProtocol, installVueDevtools } from 'vue-cli-plugin-electron-builder/lib'
const isDevelopment = process.env.NODE_ENV !== 'production'

const serialport = require('serialport')

if (isDevelopment) {
  require('module').globalPaths.push(process.env.NODE_MODULES_PATH)
}

let mainWindow

// Standard scheme must be registered before the app is ready
protocol.registerStandardSchemes(['app'], { secure: true })

function createMainWindow () {

  const window = new BrowserWindow({ kiosk: true, frame: true, icon: __dirname + '/assets/touch_icon_gbB_icon.ico'})

  if (isDevelopment) {

    // Load the url of the dev server if in development mode
    window.loadURL(process.env.WEBPACK_DEV_SERVER_URL)
    if (!process.env.IS_TEST) window.webContents.openDevTools()

  } else {

    createProtocol('app')
    //   Load the index.html when not in development
    window.loadURL(
      formatUrl({
        pathname: path.join(__dirname, 'index.html'),
        protocol: 'file',
        slashes: true
      })
    )

  }

  window.on('closed', () => {
    app.quit();
    mainWindow = null
  })

  window.webContents.on('devtools-opened', () => {
    window.focus()
    setImmediate(() => {
      window.focus()
    })
  })

  return window

}

app.on('activate', () => {
  // on macOS it is common to re-create a window even after all windows have been closed
  if (mainWindow === null) {
    mainWindow = createMainWindow()
  }
})
```

### Versions, Operating System and Hardware

- SerialPort@7.1.5
- Node.js v8.11.3
- Windows

HipsterBrown, 2019-08-17:
@Wolrab1 Thanks for creating an issue. What version of Electron are you using? This is most likely an issue because Serialport can't access any paths because it's being bundled by Webpack for the browser, which doesn't contain serialports. [electron-serialport](https://github.com/nodebots/electron-serialport) is a good example of using serialport with electron.

nklayman, 2019-08-22:
Hey, it's the creator of vue-cli-plugin-electron builder. You'll need to mark serialport as an external by setting your `vue.config.js` to:

```js
module.exports = {
  pluginOptions: {
    electronBuilder: {
      externals: ['serialport']
    }
  }
}
```

closed 2019-10-28 by stale[bot]