serialport/node-serialport #1907: node-serialport causes issues with binding module

Rushinko, 2019-08-08:
### Summary of Problem

(Please answer all 3)

I am trying to implement the node-serialport module into a react application. However, when I add `const SerialPort = require('serialport');` My application crashes saying TypeError: exist is not a function, and points to /node_modules/bindings/bindings.js

i installed serialport using `npm install serialport`  
and I have also tried `npm install serialport --build-to-source`

What can I do to fix this issue?

Thank you

### Code to Reproduce the Issue

```js
const SerialPort = require('serialport')

```

[image]

### Versions, Operating System and Hardware

- SerialPort@7
- Node.js v12.8.0
- Windows 10 (x64)

reconbot, 2019-08-08:
Are you running in a browser?

Rushinko, 2019-08-08:
> Are you running in a browser?
> -- --- Francis Gulotta wizard@roborooter.com

Yes I am.

reconbot, 2019-08-08:
Browsers don’t have serialport access. There’s a chrome experimental
feature for a new serialport spec but it’s not shipping yet and probably
won’t for a long while. Most people use node or electron.

Rushinko, 2019-08-08:
I apologize I misinterpreted the first question I think. I am using node with react so it’s rendering in chrome, however I’m still using node. Would that still not work? I also have separate issues attempting it with electron but I think that’s an electron-builder issue.

reconbot, 2019-08-08:
It usually is an electron builder issue.

The serialport object only works in the node server. You need to use a web
socket or something to get data to the browser.

Rushinko, 2019-08-09:
So I managed to get the electron-rebuild to work. However when i try and run i get this error message:
[image]

I've tried updating my node with `npm install -g n` but that didn't fix anything. I've tried removing serialport from the project and re-adding it and using electron-rebuild, still didn't work. Any ideas?

HipsterBrown, 2019-08-18:
@Rushinko What version of Electron are you using? That error usually happens when serialport/bindings (or another native binary) is compiled against a different version of Node from the version running the program.

Rushinko, 2019-08-19:
> @Rushinko What version of Electron are you using? That error usually happens when serialport/bindings (or another native binary) is compiled against a different version of Node from the version running the program.

@HipsterBrown I am using v6.0.1, how can I recompile the native binaries to the target version? I thought that was electron-rebuild's purpose.

HipsterBrown, 2019-08-20:
@Rushinko It appears this could be an issue with `electron-rebuild` and `electron` https://github.com/electron/electron-rebuild/issues/314

Rushinko, 2019-08-20:
@HipsterBrown I created a new project and migrated the code over and installed serialport, electron and electron rebuild again and it is working now.

Thank you everyone for your help

closed 2019-08-20 by Rushinko