serialport/node-serialport #1909: serialport giving the undefined in angular2

jrajauriya, 2019-08-13:
## 👉 [Please follow one of these issue templates](https://github.com/serialport/node-serialport/issues/new/choose) 👈
We want to serial port communication using existing angular2 project.
Installed the following.

npm install serialport --save
npm install @types/serialport --save

and placed 
import { SerialPort } from 'serialport' in component.ts

and
serialPort: typeof SerialPort;

now serialPort giving the undefined response.
we want to list of port.
any help thanks in advance.

HipsterBrown, 2019-08-18:
@jrajauriya Thanks for making an issue. Do you want to use serialport in the browser? Or are you building an electron app with an angular front-end UI? 

If you want to use serialport in the browser, unfortunately that is not possible with this package. It needs to run in a Node.js environment to get access to your operating system's ports. 

If you're using electron, checking out https://github.com/nodebots/electron-serialport for an example of how to use serialport in an electron project.

jrajauriya, 2019-08-19:
thanks for reply.
We want to serial port in browser or client side with angular2 project.
we tried serialport and browser-serialport but not able to access serial port.
there is any other api compatible with angular2.

HipsterBrown, 2019-08-19:
There is currently no way to access serial ports from all browsers, although there is a platform API in the works -> https://github.com/WICG/serial

Chrome has an extensions and OS API currently, not sure of the support though -> https://developer.chrome.com/apps/app_serial

The only way to work in all browsers would be to run a local webserver running node-serialport on the user's computer and communicating with your web app somehow, usually websockets. This is tricky to accomplish due to user trust.

closed 2019-08-29 by jrajauriya