serialport/node-serialport #1910: NODE_MODULE_VERSION 64. This version of Node.js requires NODE_MODULE_VERSION 73. Please try re-compiling or re-installing

DonavanMartin, 2019-08-13:
bindings.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 64. This version of Node.js requires
NODE_MODULE_VERSION 73. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).
    at process.func (electron/js2c/asar.js:155:31)
    at process.func [as dlopen] (electron/js2c/asar.js:155:31)
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:828:18)
    at Object.func (electron/js2c/asar.js:155:31)
    at Object.func [as .node] (electron/js2c/asar.js:155:31)
    at Module.load (internal/modules/cjs/loader.js:645:32)
    at Function.Module._load (internal/modules/cjs/loader.js:560:12)
    at Module.require (internal/modules/cjs/loader.js:685:19)

HipsterBrown, 2019-08-18:
Hello @DonavanMartin, thanks for making an issue. Can you reply with the version of Node.js, `node-serialport`, and `electron` you are using in this project? If you changed Node.js versions in your project, the native dependencies (like Serialport's `bindings` package) will need to be rebuilt to run effectively.

ManuPerfo, 2019-08-21:
HI @DonavanMartin ,
Same issue was with me also. It was because inside the package.json the 'serialpaort' was under devDependencies, I changed it to my 'dependencies'. and was working fine

AlexisTorres-85, 2019-09-13:
An unhandled error occurred inside electron-rebuild
gyp info it worked if it ends with ok
gyp info using node-gyp@5.0.3
gyp info using node@10.15.3 | win32 | x64
gyp ERR! find Python
gyp ERR! find Python Python is not set from command line or npm configuration
gyp ERR! find Python Python is not set from environment variable PYTHON
gyp ERR! find Python checking if "python" can be used
gyp ERR! find Python - executable path is "C:\Users\atorres\AppData\Local\Programs\Python\Python37-32\python.exe"
gyp ERR! find Python - version is "3.7.4"
gyp ERR! find Python - version is 3.7.4 - should be >=2.6.0 <3.0.0
gyp ERR! find Python - THIS VERSION OF PYTHON IS NOT SUPPORTED
gyp ERR! find Python checking if "python2" can be used
gyp ERR! find Python - "python2" is not in PATH or produced an error
gyp ERR! find Python checking if the py launcher can be used to find Python 2
gyp ERR! find Python - "py.exe" is not in PATH or produced an error
gyp ERR! find Python checking if Python is C:\Python27\python.exe
gyp ERR! find Python - "C:\Python27\python.exe" could not be run
gyp ERR! find Python checking if Python is C:\Python37\python.exe
gyp ERR! find Python - "C:\Python37\python.exe" could not be run
gyp ERR! find Python
gyp ERR! find Python **********************************************************
gyp ERR! find Python You need to install the latest version of Python 2.7.
gyp ERR! find Python Node-gyp should be able to find and use Python. If not,
gyp ERR! find Python you can try one of the following options:
gyp ERR! find Python - Use the switch --python="C:\Path\To\python.exe"
gyp ERR! find Python   (accepted by both node-gyp and npm)
gyp ERR! find Python - Set the environment variable PYTHON
gyp ERR! find Python - Set the npm configuration variable python:
gyp ERR! find Python   npm config set python "C:\Path\To\python.exe"
gyp ERR! find Python For more information consult the documentation at:
gyp ERR! find Python https://github.com/nodejs/node-gyp#installation
gyp ERR! find Python **********************************************************
gyp ERR! find Python
gyp ERR! configure error

i already installed python

feeh27, 2019-09-15:
@AlexisTorres-85 the error is that you are using Python version 3.7.4 and the version must be between> = 2.6.0 and <3.0.0, change your Python version and try again.

https://datascience.com.co/how-to-install-python-2-7-and-3-6-in-windows-10-add-python-path-281e7eae62a

feeh27, 2019-09-15:
@AlexisTorres-85 the error is that you are using Python version 3.7.4 and the version must be between >= 2.6.0 and < 3.0.0, change your Python version and try again.

https://datascience.com.co/how-to-install-python-2-7-and-3-6-in-windows-10-add-python-path-281e7eae62a

marceloch2, 2019-10-11:
I have the same problem as @AlexisTorres-85 but in OSX.

An unhandled error occurred inside electron-rebuild
gyp info it worked if it ends with ok
gyp info using node-gyp@5.0.5
gyp info using node@10.16.3 | darwin | x64
gyp info find Python using Python version 2.7.15 found at "/usr/local/opt/python@2/bin/python2.7"

And my python version is 2.7.15

lilian131, 2019-10-29:
hi @marceloch2 did you figure out ? my python is 2.7.17 too

marceloch2, 2019-10-30:
Hey friend @lilian131, yes.

Moved my Electron to 7, nodejs 12, download the binaries that are getting 404 and paste it inside /Users/YOURUSER/.npm/_prebuilds

https://github.com/serialport/node-serialport/releases

bindings-v8.0.4-electron-v75-darwin-x64.tar.gz
bindings-v8.0.4-electron-v75-linux-ia32.tar.gz
bindings-v8.0.4-electron-v75-linux-x64.tar.gz
bindings-v8.0.4-electron-v75-win32-ia32.tar.gz
bindings-v8.0.4-electron-v75-win32-x64.tar.gz

I had to do it because even with the new version for serial, electron and new node still have 404.

tangsj, 2019-11-11:
run `./node_modules/.bin/electron-rebuild`

6 warnings and 19 errors generated.
make: *** [Release/obj.target/BluetoothSerialPort/src/osx/DeviceINQ.o] Error 1
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/Users/codecook/workspace/homegit/wujie/printer/node_modules/_node-gyp@5.0.5@node-gyp/lib/build.js:194:23)
gyp ERR! stack     at ChildProcess.emit (events.js:189:13)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:248:12)
gyp ERR! System Darwin 19.0.0
gyp ERR! command "/usr/local/bin/node" "/Users/codecook/workspace/homegit/wujie/printer/node_modules/_electron-rebuild@1.8.6@electron-rebuild/node_modules/.bin/node-gyp" "rebuild" "--target=7.1.1" "--arch=x64" "--dist-url=https://electronjs.org/headers" "--build-from-source"
gyp ERR! cwd /Users/codecook/workspace/homegit/wujie/printer/node_modules/_node-bluetooth@1.2.6@node-bluetooth
gyp ERR! node -v v10.15.3
gyp ERR! node-gyp -v v5.0.5
gyp ERR! not ok

iamcam, 2019-11-15:
> Hey friend @lilian131, yes.
>
> Moved my Electron to 7, nodejs 12, download the binaries that are getting 404 and paste it inside /Users/YOURUSER/.npm/_prebuilds [...]

This doesn't seem to help... though the build process chooses the wrong version of node (chooses 72 instead of 75)

mhabib555, 2019-12-06:
I was facing the same error
bindings.node was compiled against a different Node.js version using NODE_MODULE_VERSION 72. This version of Node.js requires NODE_MODULE_VERSION 75.

On windows 7, using electron 7. I solved it by using electron-builder.

My scripts in  package.json
"scripts": {
		"ng": "ng",
		"start": "ng serve",
		"build": "ng build",
		"test": "ng test",
		"lint": "ng lint",
		"e2e": "ng e2e",
		"electron": "ng build --base-href ./ && tsc --p electron && electron .",
		"pack": "electron-builder --dir",
		"dist": "electron-builder"
	},

Running "npm run dist" solved the issue.

AlexandreMouyen, 2019-12-08:
@mhabib555 Can you please show your all package.json please. This command does not work for me. Thanks

AlexandreMouyen, 2019-12-08:
I think this is the problem

prebuild-install` WARN install prebuilt binaries enforced with --force!
    prebuild-install WARN install prebuilt binaries may be out of date!
    prebuild-install info looking for cached prebuild @ C:\Users\gegem\AppData\Roaming\npm-cache\_prebuilds\8de1b9-bindings-v2.0.8-electron-v75-win32-x64.tar.gz
    prebuild-install http request GET https://github.com/node-serialport/node-serialport/releases/download/v2.0.8/bindings-v2.0.8-electron-v75-win32-x64.tar.gz
    prebuild-install http 404 https://github.com/node-serialport/node-serialport/releases/download/v2.0.8/bindings-v2.0.8-electron-v75-win32-x64.tar.gz
    prebuild-install WARN install No prebuilt binaries found (target=7.1.3 runtime=electron arch=x64 libc= platform=win32)

  ⨯ cannot execute  cause=exit status 1
                    out=
    > @serialport/bindings@2.0.8 install D:\Documents\space-controller\node_modules\firmata\node_modules\@serialport\bindings
    > prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild

mhabib555, 2019-12-08:
@AlexandreMouyen 
here is my package.json file
`{
	"version": "0.0.3",
	"main": "electron/dist/main.js",
	"scripts": {
		"ng": "ng",
		"start": "ng serve",
		"build": "ng build",
		"test": "ng test",
		"lint": "ng lint",
		"e2e": "ng e2e",
		"electron": "ng build --base-href ./ && tsc --p electron && electron .",
		"pack": "electron-builder --dir",
		"dist": "electron-builder"
	},
	"private": true,
	"dependencies": {
		"@angular/animations": "~8.2.13",
		"@angular/common": "~8.2.13",
		"@angular/compiler": "~8.2.13",
		"@angular/core": "~8.2.13",
		"@angular/forms": "~8.2.13",
		"@angular/platform-browser": "~8.2.13",
		"@angular/platform-browser-dynamic": "~8.2.13",
		"@angular/router": "~8.2.13",
		"modem": "github:apla/modem",
		"ngx-electron": "^2.2.0",
		"node-adodb": "^5.0.2",
		"rxjs": "~6.4.0",
		"tslib": "^1.10.0",
		"zone.js": "~0.9.1"
	},
	"devDependencies": {
		"@angular-devkit/build-angular": "~0.803.18",
		"@angular/cli": "~8.3.18",
		"@angular/compiler-cli": "~8.2.13",
		"@angular/language-service": "~8.2.13",
		"@types/jasmine": "~3.3.8",
		"@types/jasminewd2": "~2.0.3",
		"@types/node": "~8.9.4",
		"codelyzer": "^5.0.0",
		"electron": "^7.1.1",
		"electron-builder": "^22.1.0",
		"jasmine-core": "~3.4.0",
		"jasmine-spec-reporter": "~4.2.1",
		"karma": "~4.1.0",
		"karma-chrome-launcher": "~2.2.0",
		"karma-coverage-istanbul-reporter": "~2.0.1",
		"karma-jasmine": "~2.0.1",
		"karma-jasmine-html-reporter": "^1.4.0",
		"protractor": "~5.4.0",
		"ts-node": "~7.0.0",
		"tslint": "~5.15.0",
		"typescript": "~3.5.3"
	},
	"build": {
		"appId": "***.***.***",
		"files": [
			"electron/dist/main.js",
			"build"
		],
		"win": {
			"target": "portable",
			"icon": "electron/assets/icons/icon.ico"
		},
		"extraResources": [
		]
	}
}
`

You can also  check this.
https://mhabib555.wordpress.com/2019/12/06/nodejs-serialport-gsm-building-error-on-electron-using-electron-builder-npm-run-dist/

Damilare1, 2020-01-12:
> bindings.node'
> was compiled against a different Node.js version using
> NODE_MODULE_VERSION 64. This version of Node.js requires
> NODE_MODULE_VERSION 73. Please try re-compiling or re-installing
> the [...]

**You may have to upgrade your node version. npm -i only updates to the latest stable LTS version which may still have a NODE_MODULE_VERSION < 73. Checkout [(https://nodejs.org/en/download/releases/#ref-1)]**

Damilare1, 2020-01-12:
You may have to upgrade your node version. npm -i only updates to the latest stable LTS version which may still have a NODE_MODULE_VERSION < 73. Checkout [(https://nodejs.org/en/download/releases/#ref-1)]

closed 2020-03-19 by stale[bot]

carboleda, 2020-07-26:
> Hi @DonavanMartin,
> 
> I solved this problem using the command below:
> 
> ```shell
> ./node_modules/.bin/electron-rebuild
> ```
> 
> **Steps:**
> 
> * If you have not installed electron-rebuild just install it with the command: `npm i -D electron-rebuild`
> * Remove from the `node-modules` folder the `serialport` and ` @serialport` folders.
> * Remove the file `packages-lock.json`
> * Run `npm i` to install non-installed modules
> * And finally run `./node_modules/.bin/electron-rebuild`
> 
> It is very important to run `./node_modules/.bin/electron-rebuild` directly after `npm i`.
> 
> Source: https://stackoverflow.com/a/52796884

Thank you, this work for me!

Ladvien, 2020-08-10:
@feeh27 instructions did the trick for me.  However, when using electron-rebuild I had to specify the version--for me, was trying to make VSCode.
E.g.,
```
./node_modules/.bin/electron-rebuild -v 7
```

CarlNFT, 2021-12-09:
hello 
I have that issue 

Error: The module '/Users/carlito/Downloads/hashlips_art_engine-main 2/node_modules/canvas/build/Release/canvas.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 64. This version of Node.js requires
NODE_MODULE_VERSION 93. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).
    at Object.Module._extensions..node (node:internal/modules/cjs/loader:1183:18)
    at Module.load (node:internal/modules/cjs/loader:981:32)
    at Function.Module._load (node:internal/modules/cjs/loader:822:12)
    at Module.require (node:internal/modules/cjs/loader:1005:19)
    at require (node:internal/modules/cjs/helpers:102:18)
    at Object.<anonymous> (/Users/carlito/Downloads/hashlips_art_engine-main 2/node_modules/canvas/lib/bindings.js:3:18)
    at Module._compile (node:internal/modules/cjs/loader:1101:14)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1153:10)
    at Module.load (node:internal/modules/cjs/loader:981:32)
    at Function.Module._load (node:internal/modules/cjs/loader:822:12) {
  code: 'ERR_DLOPEN_FAILED'
}
carlito@iMac hashlips_art_engine-main 2 % 

Thanks for your help

GazHank, 2021-12-09:
Hi @CarlNFT are you using electron?  If so the instructions in the earlier comments should help you

TzachiGuetta, 2022-12-04:
worked for me:
cd /opt/zigbee2mqtt/
rm -rf node-modules/serialport
rm -rf node-modules/@serialport
rm -rf packages-lock.json
sudo npm i
sudo systemctl start zigbee2mqtt

yogithesymbian, 2023-09-02:
have same issue but different package
```
Error: The module '/var/www/api/node_modules/canvas/build/Release/canvas.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 64. This version of Node.js requires
NODE_MODULE_VERSION 108. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).
```