serialport/node-serialport #1918: Setup Flow Control option to none in port open configuration

mumairofficial, 2019-08-20:
### Summary of Problem

(Please answer all 3)

- What are you trying to do?
Configuring the port according to putty settings as follows.

Need help to setup **Flow control** value to **'none'**

[image]

- What happens?
Data write works perfect but sometime data receive did not work

- What should have happened?
It should receive data

### Code to Reproduce the Issue

```js
const serialport = new SerialPort(path)

serialport = new serialport(comPort, {
    baudRate: 115200,
    autoOpen: false,
    dataBits: 8,
    stopBits: 1,
    parity: 'none',
  
    // how to setup flow control to none; there no such option in OpenOptions interface
    // following is my assumption
    xon: false,
    xoff: false
  })
```

### Versions, Operating System and Hardware

- SerialPort@6.2.1
- Node.js v10.x
- Windows & Linux
- Hardware and chipset? (Prolific, 9 Pin Serial cable)

NorbarTDB, 2019-09-06:
Just floated across this issue while looking for documentation on flow control! Seems like there isn't really any, sadly.

I'm not sure myself (hence the looking), but I wonder if this would work:

```javascript
serialport = new serialport(comPort, {
    baudRate: 115200,
    autoOpen: false,
    dataBits: 8,
    stopBits: 1,
    parity: 'none',
  
 
    xon: false,
    xoff: false,
    rtscts: false
  })
```

Though, when I create a port, this seems to be the default configuration anyway? At least, if I don't specify `rtscts` on opening a port, it seems to wind up set to false anyway:

[image]

xenkuo, 2020-02-13:
There are types of flow control, but normally it means hardware flow control, which uses cts/rts pins. For software flow control, it uses `xany` `xon` and `xoff`. 
Both above flow control are disabled if you did not set them when new port.
So I don't understand why add this to a new feature request, did I miss something?

GazHank, 2021-07-13:
Should this be resolved through an update to the docs rather than a code change, or is there some additional functionality we need to add/change?