serialport/node-serialport #1919: npm fails to install @serialport/bindings

mraghup, 2019-08-22:
### Summary of Problem

(Please answer all 3)

- What are you trying to do?
- What happens?
- What should have happened?

### Code to Reproduce the Issue

```js
const serialport = new SerialPort(path)

// Code
```

### Versions, Operating System and Hardware

- SerialPort@?
- Node.js v?
- Windows? Linux? Mac?
- Hardware and chipset? (Prolific/FTDI/Other)

mraghup, 2019-08-22:
Looks like the issue may have been due to the fact out Win Server 2016 uses npm version 6.11.0 where as our WIn 10 system where npm build serialport successfully is using npm 6.9.0. After downgrading npm from 6.11.0 to 6.9.0 the native add on builds without issues.

reconbot, 2019-08-30:
npm 6.11.0 works on other platforms - this looks maybe like a prebuild issue that they fixed while you were testing? Or maybe an issue with the paths on the other computer? Different versions of npm wouldn't affect what gets installed via npm.

closed 2019-08-30 by reconbot