serialport/node-serialport #1920: Some published Prebuild versions have incorrect ABI

Josverl, 2019-08-24:
### Summary of Problem
Im trying to use the prebuild bindaries that are published to provide support for : Multiple plaforms, CPU architectures and versions of electron.

This fails as the ABI versions of the published prebuilts  do not match.

Note: this is not regarding the match to a users current runtime
Rather there appears to be something incorrect in that 
- for the published version ABI 70 for electron 
- the files themseve inform NodeJS that they are 47
and thus they cannot be loaded.

Ive found is specifically in the below files , but there may be others as well 

| File abi | file | runtime|
|---------|----|------------|
| 47 | electron-v70-win32-x64\bindings.node| electron |
| 47 | electron-v70-linux-x64/bindings.node | electron |

### Code to Reproduce the Issue

- download the prebuild for electron 5

```
npx prebuild-install --runtime electron --target 5.0.10 --arch x64 --platform win32 --tag-prefix @serialport/bindings@ --verbose
prebuild-install info begin Prebuild-install version 5.3.0
prebuild-install info looking for cached prebuild @ C:\Users\josverl\AppData\Roaming\npm-cache\_prebuilds\bde028-bindings-v2.0.8-electron-v70-win32-x64.tar.gz
prebuild-install http request GET https://github.com/node-serialport/node-serialport/releases/download/@serialport/bindings@2.0.8/bindings-v2.0.8-electron-v70-win32-x64.tar.gz
prebuild-install http 200 https://github.com/node-serialport/node-serialport/releases/download/@serialport/bindings@2.0.8/bindings-v2.0.8-electron-v70-win32-x64.tar.gz
prebuild-install info downloading to @ C:\Users\josverl\AppData\Roaming\npm-cache\_prebuilds\bde028-bindings-v2.0.8-electron-v70-win32-x64.tar.gz.16608-da68a1f0e84a4.tmp
prebuild-install info renaming to @ C:\Users\josverl\AppData\Roaming\npm-cache\_prebuilds\bde028-bindings-v2.0.8-electron-v70-win32-x64.tar.gz
prebuild-install info unpacking @ C:\Users\josverl\AppData\Roaming\npm-cache\_prebuilds\bde028-bindings-v2.0.8-electron-v70-win32-x64.tar.gz
prebuild-install info unpack resolved to C:\develop\NodeJS\electron-serialport\node_modules\@serialport\bindings\build\Release\bindings.node
prebuild-install info install Successfully installed prebuilt binary!
VERBOSE: Performing the operation "Copy File" on target "Item: C:\develop\NodeJS\electron-serialport\node_modules\@serialport\bindings\build\Release\bindings.node Destination: C:\develop\NodeJS\electron-serialport\native_modules\@serialport\bindings\lib\binding\node-v70-win32-x64\bindings.node".

then copy native_modules/* --> node_modules 

```
- try to load it 

``` js
const binding= require("node_modules\@serialport\bindings\lib\binding\node-v70-win32-x64\bindings.node")

// or 
const serial = require("serialport")
// internally this uses: 
// bindings = require("bindings")("binding") 
// to try to load the same module
```

this will throw "
```
Uncaught Error: The module '\\?\C:\develop\NodeJS\electron-serialport\node_modules\@serialport\bindings\lib\binding\node-v70-win32-x64\bindings.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 47. This version of Node.js requires
NODE_MODULE_VERSION 70. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).
```

also tried by directly downloading the binding from the published bindings,
same problem

the issue also replication on Linux x64, 
i have not tested on Mac or on ia32 architectures.

## relevant stuff

[CheckABI Script](https://gist.github.com/Josverl/ccaf7f93dabc3fa8fc3088d058efd485) to simplify testing of ABI versions of Modules 

[check-ubuntu-x64.txt](https://github.com/serialport/node-serialport/files/3537488/check-ubuntu-x64.txt)
[check-wi32-nx64.txt](https://github.com/serialport/node-serialport/files/3537489/check-wi32-nx64.txt)

### Versions, Operating System and Hardware
* Windows 10, x64 
* Ubuntu 18.04 , x64 

- "serialport": "^7.1.5"
- Electron version 5.x.x / ABI 70

almashi92, 2019-08-26:
I have a same problem with electron version 5.0.6 on debian 9 x64.

Josverl, 2019-08-26:
Hi almas hi,
I would expect that for all electron 5.x.x
Can you confirm if you are using 32 or 64 bits?

I can also confirm the same issue on win32_ia32

almashi92, 2019-08-27:
64bits, today i tried to rebuild with electron v 2.0.8 and everything works.

Josverl, 2019-08-27:
yes, electron-rebuild is the work-around.
the point is that rebuilding should not be needed for the published prebuilts, right ?
why bother publishing them if they only throw unclear errors and make users look for problems that they have not created themselves.

I spend way to long getting to the bottom of this , and I want to prevent others from needing to do the same grunt-work

Josverl, 2019-08-27:
I have not done a full sweep , and I also noticed that some of the electron 2.x.x and 3.x.x fail to load with other error codes in the `electron-test` application that I have adopted to use as a testing ground to figure this one out.

I've not spend the time to dive into your build process, but I would say that it would make sense to figure out if this is in the build - or in the publication - of the release that things get mixed up / bad.

Josverl, 2019-08-27:
@reconbot, 
do you know if the newer bindings are in a better state than the bindings@2.0.8 ?

reconbot, 2019-08-27:
No idea, the process which has worked for a long while hasn’t changed much.

Josverl, 2019-08-30:
similar issue : https://github.com/serialport/node-serialport/issues/1884

phhoef, 2019-09-15:
I've the exact same issue on macOS (darwin), but `electron-rebuild` does not fix the problem for me.

Josverl, 2019-09-18:
well , im dont know [mac](https://www.bing.com/videos/search?q=mac+book+commercial&view=detail&mid=A41721E0A8339A051B2AA41721E0A8339A051B2A&FORM=VIRE) that well , but there is more than I know how to fix ;-)

phhoef, 2019-09-18:
hehe ;-) 

From my point of view, the root problem is still the wrong ABI version of the prebuilds of serialport ,right?
@reconbot Did you have the chance to look into that problem?

reconbot, 2019-09-18:
I’m hoping the problem was fixed upstream. Prebuild is very popular and we’re on the latest version. I just pushed serialport 8.0.1 which has new builds. Give it a go?

phhoef, 2019-09-18:
Ah, thanks. I didn't recognize your update!
I changed the `serialport` version to 8.0.1 and issued `yarn`.
Unfortunately, I doesn't work for me. I tested it with Electron 5 and 6 and getting the following error again:
```
was compiled against a different Node.js version using
NODE_MODULE_VERSION 48. This version of Node.js requires
NODE_MODULE_VERSION 73. Please try re-compiling or re-installing
```
respectively for Electron 5
```
was compiled against a different Node.js version using
NODE_MODULE_VERSION 47. This version of Node.js requires
NODE_MODULE_VERSION 70. Please try re-compiling or re-installing
```

I see, that the prebuilds are downloaded and put into the cache folder.
```
➜  _prebuilds ls -la
total 120
drwxr-xr-x   5 ph  staff    160 18 Sep 14:09 .
drwxr-xr-x  12 ph  staff    384 18 Sep 14:04 ..
-rw-r--r--   1 ph  staff  18915 18 Sep 14:04 261148-bindings-v8.0.1-electron-v70-darwin-x64.tar.gz
-rw-r--r--   1 ph  staff  19541 18 Sep 14:09 503a78-bindings-v8.0.1-node-v72-darwin-x64.tar.gz
-rw-r--r--   1 ph  staff  18916 18 Sep 14:06 5d0248-bindings-v8.0.1-electron-v73-darwin-x64.tar.gz
```

So, I think this is ok, despite I am getting this message:
```
build native dependency from sources  name=@serialport/bindings
                                          version=8.0.1
                                          platform=darwin
                                          arch=x64
                                          reason=prebuild-install failed with error (run with env DEBUG=electron-builder to get more information)
                                          error=prebuild-install info begin Prebuild-install version 5.3.2
    prebuild-install info looking for cached prebuild @ /Users/ph/.npm/_prebuilds/b4311d-bindings-v8.0.1-electron-v70-darwin-x64.tar.gz
    prebuild-install http request GET https://github.com/serialport/node-serialport/releases/download/v8.0.1/bindings-v8.0.1-electron-v70-darwin-x64.tar.gz
    prebuild-install http 404 https://github.com/serialport/node-serialport/releases/download/v8.0.1/bindings-v8.0.1-electron-v70-darwin-x64.tar.gz
    prebuild-install WARN install No prebuilt binaries found (target=5.0.10 runtime=electron arch=x64 libc= platform=darwin)
```

phhoef, 2019-09-29:
Tested it with serial port 8.0.2 and still have the same issue.
Not sure, if I am doing sth wrong?

```
/Users/ph/Documents/Projects/Client/node_modules/bindings/bindings.js:121 Uncaught Error: The module '/Users/ph/Documents/Projects/Client/node_modules/@serialport/bindings/build/Release/bindings.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 48. This version of Node.js requires
NODE_MODULE_VERSION 73. Please try re-compiling or re-installing
```

phhoef, 2019-10-04:
`8.0.3` fixes the problem for me :-)
Thanks!

Josverl, 2019-10-04:
@phhoef, thabks for you perseverance and report
 - what platforms / runtimes and bitnessses were you able to validate ?

@reconbot, did you change anything ( i did not see anything obvious) , or was there an upstream fix that you are aware of ?

phhoef, 2019-10-04:
I am developing on macOS Mojave (10.14.6) and building against macOS and Windows both x86_64.

So, the prebuilds I am using are:
* bindings-v8.0.3-electron-v73-darwin-x64.tar.gz
* bindings-v8.0.3-electron-v73-win32-x64.tar.gz

reconbot, 2019-10-05:
I have not changed anything but I haven’t looked upstream. We don’t use the
lock file in builds and testing so we might have gotten a patch update.

reconbot, 2019-10-06:
I've removed the following builds for `@serialport/bindings@2.0.8`
- electron-v70-win32-x64
- electron-v70-win32-ia32
- electron-v70-linux-x64

reconbot, 2019-10-10:
Let me know if you find any others.

closed 2019-10-10 by reconbot