serialport/node-serialport #1929: npx @serialport/list crashes on Windows

AlCalzone, 2019-09-03:
### Summary of Problem

(Please answer all 3)

- What are you trying to do?
--> Execute `npx @serialport/list`
- What happens?
```
$ npx @serialport/list
COM3    USB\VID_0658&PID_0200\6&32BF5478&0&5    Microsoft
fs.js:125
    throw err;
    ^

Error: EPERM: operation not permitted, unlink 'C:\Users\domin\AppData\Roaming\npm-cache\_npx\20988\node_modules\@serialport\list\node_modules\@serialport\bindings\build\Release\bindings.node'
    at Object.unlinkSync (fs.js:1009:3)
    at C:\Users\domin\AppData\Roaming\npm\node_modules\npm\node_modules\rimraf\rimraf.js:342:5
    at Array.forEach (<anonymous>)
    at rmkidsSync (C:\Users\domin\AppData\Roaming\npm\node_modules\npm\node_modules\rimraf\rimraf.js:341:26)
    at rmdirSync (C:\Users\domin\AppData\Roaming\npm\node_modules\npm\node_modules\rimraf\rimraf.js:334:7)
    at fixWinEPERMSync (C:\Users\domin\AppData\Roaming\npm\node_modules\npm\node_modules\rimraf\rimraf.js:209:5)
    at rimrafSync (C:\Users\domin\AppData\Roaming\npm\node_modules\npm\node_modules\rimraf\rimraf.js:311:28)
    at C:\Users\domin\AppData\Roaming\npm\node_modules\npm\node_modules\rimraf\rimraf.js:342:5
```
- What should have happened?
```
$ npx @serialport/list
COM3    USB\VID_0658&PID_0200\6&32BF5478&0&5    Microsoft
```

### Code to Reproduce the Issue

```bash
npx @serialport/list
```

### Versions, Operating System and Hardware

- @serialport/list@1.1.5
- Node.js v12.1.0
- Windows
- Hardware and chipset? (Prolific/FTDI/Other) Not sure?!

reconbot, 2019-09-03:
I’m willing to bet this is an npx issue. We do delete any files or anything.

AlCalzone, 2019-09-04:
Could it be that npx tries to delete the compiled binary but that is still in use?

closed 2019-11-10 by stale[bot]