serialport/node-serialport #1931: Response not the same as in test-terminal program

toineKeistoer, 2019-09-05:
### Summary of Problem
Hi all, I'm trying to read an ECU device, but I'm not sure in which format to send the commands to the device. I have a helper terminal program (https://sites.google.com/site/terminalbpp/), which has some macro's that are working. 

An example of a macro is:
```
LogOn
$86$01%SUM00
```

When I use this "LogOn" message in the terminal I get a lot of data output, but when I send this message to the port with node-serialport I only receive `4646464646`:

[image]

I am doubting if I can send the same message to the port in node as in the terminal. Does anyone know this should work the same? 

Many thanks!

### Code to Reproduce the Issue

```js
const SerialPort = require('serialport');
const port = new SerialPort(
    'COM3',
    {
        baudRate: 921600,
        dataBits: 8,
        stopBits: 1,
        parity: 'space',
        lock: true,
        autoOpen: true
    },
    function(err) {
        if (err) {
            return console.log('Error: ', err.message);
        }
    }
);

port.on('data', function(data) {
    console.log('port on data');
    console.log('Data-buffer:', data);
    console.log('Data-string:', data.toString('HEX'));
});

port.on('open', function() {
    console.log('port on open');
    port.write('LogOn');
});
```

### Versions, Operating System and Hardware

- SerialPort v7.1.5
- Node.js v10.16.3
- Windows 10

reconbot, 2019-09-05:
What hardware are you using?

toineKeistoer, 2019-09-06:
Its a custom-made RS232 connection. I think I can narrow my questions down to:

1. How can I translate the commando "$86$01%SUM00" that terminalbbp sends to this node-serialport version.
2. Do I need to configure something to keep the connection open, so the device can keep sending data messages?

Thanks!

reconbot, 2019-09-06:
Unless the devise is supposed to send ascii `ffffff` I wonder if you have the right baud and communication settings.

To answer your questions:
- I don’t really understand what the macro is, you’re writing the ascii string logon with no trailing new lines or anything, when a device takes ascii they usually look for new lines to know you’ve finished typing.
- As long as the port is open you’ll get data.

toineKeistoer, 2019-09-09:
It works, I needed to send the macro command in HEX instead of the string: 
```
port.write(Buffer.from([0x86, 0x01, 0x87]));
```
The last is a sum checksum.

This topic can be closed, thanks!

closed 2019-09-09 by toineKeistoer