serialport/node-serialport #1932: Compile errors with Node.js 13 nightly build

fivdi, 2019-09-08:
### Summary of Problem

- What are you trying to do? Install serialport@7.1.5 with [Node.js 13.0.0-nightly20190908e585caa2be](https://nodejs.org/download/nightly/v13.0.0-nightly20190908e585caa2be/)
- What happens? The installation terminates with compile errors
- What should have happened? serialport should have installed without compile errors

### Code to Reproduce the Issue

```
npm i serialport
```

### Versions, Operating System and Hardware

- SerialPort@? 7.1.5
- Node.js v? v13.0.0-nightly20190908e585caa2be
- Windows? Linux? Mac? Linux (Debian 10, gcc 8.3.0)
- Hardware and chipset? (Prolific/FTDI/Other) BeagleBone Black (ARMv7)

### Output of `npm i serialport`

```
debian@beaglebone:~/serialport$ node -v
v13.0.0-nightly20190908e585caa2be

debian@beaglebone:~/serialport$ gcc --version
gcc (Debian 8.3.0-6) 8.3.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

debian@beaglebone:~/serialport$ npm i serialport
npm WARN npm npm does not support Node.js v13.0.0-nightly20190908e585caa2be
npm WARN npm You should probably upgrade to a newer version of node as we
npm WARN npm can't make any promises that npm will work with this version.
npm WARN npm Supported releases of Node.js are the latest release of 6, 8, 9, 10, 11, 12.
npm WARN npm You can find the latest version at https://nodejs.org/

> @serialport/bindings@2.0.8 install /home/debian/serialport/node_modules/@serialport/bindings
> prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild

prebuild-install WARN install No prebuilt binaries found (target=13.0.0-nightly20190908e585caa2be runtime=node arch=arm libc= platform=linux)
make: Entering directory '/home/debian/serialport/node_modules/@serialport/bindings/build'
  CXX(target) Release/obj.target/bindings/src/serialport.o
In file included from ../src/./serialport.h:6,
                 from ../src/serialport.cpp:1:
../../../nan/nan.h: In function ‘void Nan::AsyncQueueWorker(Nan::AsyncWorker*)’:
../../../nan/nan.h:2298:62: warning: cast between incompatible function types from ‘void (*)(uv_work_t*)’ {aka ‘void (*)(uv_work_s*)’} to ‘uv_after_work_cb’ {aka ‘void (*)(uv_work_s*, int)’} [-Wcast-function-type]
     , reinterpret_cast<uv_after_work_cb>(AsyncExecuteComplete)
                                                              ^
../src/serialport.cpp: In function ‘Nan::NAN_METHOD_RETURN_TYPE Open(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/serialport.cpp:78:69: warning: cast between incompatible function types from ‘void (*)(uv_work_t*)’ {aka ‘void (*)(uv_work_s*)’} to ‘uv_after_work_cb’ {aka ‘void (*)(uv_work_s*, int)’} [-Wcast-function-type]
   uv_queue_work(uv_default_loop(), req, EIO_Open, (uv_after_work_cb)EIO_AfterOpen);
                                                                     ^~~~~~~~~~~~~
../src/serialport.cpp: In function ‘Nan::NAN_METHOD_RETURN_TYPE Update(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/serialport.cpp:135:71: warning: cast between incompatible function types from ‘void (*)(uv_work_t*)’ {aka ‘void (*)(uv_work_s*)’} to ‘uv_after_work_cb’ {aka ‘void (*)(uv_work_s*, int)’} [-Wcast-function-type]
 uv_queue_work(uv_default_loop(), req, EIO_Update, (uv_after_work_cb)EIO_AfterUpdate);
                                                                     ^~~~~~~~~~~~~~~

../src/serialport.cpp: In function ‘Nan::NAN_METHOD_RETURN_TYPE Close(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/serialport.cpp:175:70: warning: cast between incompatible function types from ‘void (*)(uv_work_t*)’ {aka ‘void (*)(uv_work_s*)’} to ‘uv_after_work_cb’ {aka ‘void (*)(uv_work_s*, int)’} [-Wcast-function-type]
  uv_queue_work(uv_default_loop(), req, EIO_Close, (uv_after_work_cb)EIO_AfterClose);
                                                                     ^~~~~~~~~~~~~~

../src/serialport.cpp: In function ‘Nan::NAN_METHOD_RETURN_TYPE Flush(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/serialport.cpp:215:70: warning: cast between incompatible function types from ‘void (*)(uv_work_t*)’ {aka ‘void (*)(uv_work_s*)’} to ‘uv_after_work_cb’ {aka ‘void (*)(uv_work_s*, int)’} [-Wcast-function-type]
  uv_queue_work(uv_default_loop(), req, EIO_Flush, (uv_after_work_cb)EIO_AfterFlush);
                                                                     ^~~~~~~~~~~~~~

../src/serialport.cpp: In function ‘Nan::NAN_METHOD_RETURN_TYPE Set(Nan::NAN_METHOD_ARGS_TYPE)’:
[... 121 lines cut ...]
gyp ERR! System Linux 4.19.50-ti-r20
gyp ERR! command "/home/debian/opt/node-v13.0.0-nightly20190908e585caa2be-linux-armv7l/bin/node" "/home/debian/opt/node-v13.0.0-nightly20190908e585caa2be-linux-armv7l/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/debian/serialport/node_modules/@serialport/bindings
gyp ERR! node -v v13.0.0-nightly20190908e585caa2be
gyp ERR! node-gyp -v v5.0.3
gyp ERR! not ok 
npm WARN enoent ENOENT: no such file or directory, open '/home/debian/serialport/package.json'
npm WARN serialport No description
npm WARN serialport No repository field.
npm WARN serialport No README data
npm WARN serialport No license field.
npm WARN You are using a pre-release version of node and things may not work as expected

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! @serialport/bindings@2.0.8 install: `prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the @serialport/bindings@2.0.8 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/debian/.npm/_logs/2019-09-08T14_55_59_932Z-debug.log
debian@beaglebone:~/serialport$ 
```

reconbot, 2019-09-08:
Can you build master? We landed a fix for this I think

fivdi, 2019-09-08:
Yep, master builds successfully. It was fixed with #1901.

closed 2019-09-08 by fivdi