serialport/node-serialport #1934: port.write sends incomplete data

wasitum, 2019-09-13:
### Summary of Problem

- What are you trying to do?
I am trying to send a string to a serialport that will print it on a thermal printer

- What happens?
I am sending the string to the port, but only the 10 first characters are being received.

- What should have happened?
All characters in the string should have been sent.
When I sent it through a terminal using command :
```sh
 screen -L /dev/tty.usbserial-DN03U84D 115200
```
 and then typing it and pressing Enter, all characters are received

Example :
String being sent : 41iabcdefghijklmnopqrst\r
Printer shows : 41iabcdefg
Sending the same string through the terminal, printer shows : 41iabcdefghijklmnopqrst

### Code to Reproduce the Issue

```js
const SerialPort = require("serialport");

const baudRate = 115200;
const comPort = "/dev/tty.usbserial-DN03U84D";
const lock = true;

const device = new SerialPort(comPort, { baudRate, lock });

device.write('41iabcdefghijklmnopqrst\r');

```

### Versions, Operating System and Hardware

- SerialPort@7.1.5
- Node.js v10.0.0
- OSX Mojave

reconbot, 2019-09-13:
does this script quit automatically? And are you calling drain anywhere?

wasitum, 2019-09-23:
@reconbot In this example code yes the code is quitting automatically, but the general code stays up. I have an array of commands to be executed, the code execute the first one every second, and stays up as long as we don't quit the execution.

```js
setInterval(() => {
  if (queue.length) {
    const instruction = queue.splice(0, 1);
    console.log("Exécution : " + instruction);
    device.write(instruction[0] + '\r', 'utf8', function (err) {
      if (err) {
        return console.log('Error on write: ', err.message)
      }
    });
    device.drain();
  }
}, 1000);
```

boatincow, 2019-10-30:
Same problem, both on MacOS and Windows some symbols randomly missing

closed 2020-01-06 by stale[bot]

saschagehlich, 2020-11-12:
@wasitum I'm currently experiencing a similar issue. Did you find a solution?