serialport/node-serialport #1939: Infinite callback loop in unix-read.js (7.1.5)

AndrolGenhald, 2019-09-24:
### Summary of Problem

- What are you trying to do?

Read from a serial port which another process may also be trying to read from. This itself is the actual issue of course, but I believe it may have exposed underlying behavior that should still be fixed.

- What happens?

node-serialport causes an infinite callback loop, which runs nodejs out of memory and causes it to crash.

Due to this code in unix-read.js:
```
if (bytesRead === 0) {
  resolve(this.read(buffer, offset, length))
  return
}
```
If another process reads all the data from the serial port before node-serialport can, it will keep reading 0 bytes and try again indefinitely, eventually running out of memory from all of the `this.read` calls.

- What should have happened?

The infinite callback loop should be avoided. Perhaps an error should be thrown instead if data was expected but unable to be read.

### Code to Reproduce the Issue

Reproduction is dependent on external factors, namely another process reading from the serial port.

### Versions, Operating System and Hardware

- SerialPort@?

7.1.5

This is no longer an issue in 8.0.2, so feel free to close this if 7.1.5 is no longer supported.

- Node.js v?

12.4.0

- Windows? Linux? Mac?

Linux

- Hardware and chipset? (Prolific/FTDI/Other)

Not really sure...but it's a raspberry pi if that tells you anything.

reconbot, 2019-09-25:
If it's fixed in the 8x beta then we're in good shape. We don't usually backport bug fixes.

Plasmarobo, 2019-12-06:
This seems to still exist in 8.0.5:
in unix-read.js:
```
const { bytesRead } = await readAsync(this.fd, buffer, offset, length, null)
    if (bytesRead === 0) {
      return this.read(buffer, offset, length)
    }
```
This creates an infinite loop if the port isn't currently returning data, putting a ton of read calls on the heap until it crashes.
I would expect this to resolve each step, without making them depend on the resolution of previous or future steps. I'm not sure what steps could be taken to fix it, and I'm working on a workaround.

reconbot, 2019-12-25:
I'm a little conflicted on what to do. The [`read(2)` man page](https://linux.die.net/man/2/read) specifies read only returns 0 bytes when the "file" is ended.  Which mean the port is closed. In the other cases it should return -1 and set the `EAGAIN` error. Which causes us to poll for a readable state. 

In your case, you're getting 0 bytes but not `EAGAIN` so I'm not sure if we'd ever get another readable event for us to EPOLL via the `poller` class. I'm inclined to close the port, but in @AndrolGenhald case this isn't desireable.

The current state of this code is as follows;
```js
    const { bytesRead } = await fsReadAsync(binding.fd, buffer, offset, length, null)
    if (bytesRead === 0) {
      return unixRead({ binding, buffer, offset, length, fsReadAsync })
    }
```

This causes a promise chain and not something that can blow the stack like recursive sync functions. 

@Plasmarobo do you have debug logs for your use case?

AndrolGenhald, 2019-12-25:
Just to be clear, I'm ok with closing the port if 0 bytes are read, having multiple processes reading was more of an accident than a legitimate use case. I don't think it's really unreasonable to say nothing else should be reading from the port at the same time, and closing in a detectable way is certainly better than blowing the stack like it used to.

Plasmarobo, 2019-12-26:
Wouldn't 0 bytes also be returned if we've read all _currently available_ bytes in a stream?
I'm not sure why it's not emitting EAGAIN or EWOULDBLOCK. Perhaps the fd is not marked nonblocking?
Wondering if it's a filesystem issue: I'm opening a symlink which follows a device-id, not sure how that interacts with the read() file descriptors.

I can try to gather debug logs the next time I dive into this. What specifically would be helpful to log?
My setup has some support for remote debugging via ndb/chrome/vscode, shouldn't be too much trouble.

reconbot, 2019-12-26:
@AndrolGenhald
>  I'm ok with closing the port if 0 bytes are read, having multiple processes reading was more of an accident than a legitimate use case.
That's good to know.

@Plasmarobo We use [debug](https://www.npmjs.com/package/debug) you can find more info [here about how to enable it for serialport](https://serialport.io/docs/guide-debugging). I doubt the symlink would be an issue but it's worth a try.

I don't know why this would blow the call stack. Its possible we had a memory leak or something? I'd be curious to know more about what you're seeing in the current version of serialport. The JavaScript with recursive async functions wont blow the stack so to speak.

For example, this is safe.
```js
const fakeAsyncRead = () => new Promise(resolve => setImmediate(resolve))

let count = 1
const foo = async () => {
  await fakeAsyncRead()
  console.log(count++)
  return foo()
}

foo()
```

This however is not
```js
let count = 1
const foo = async () => {
  console.log(count++)
  return foo()
}
```

closed 2020-03-03 by stale[bot]