serialport/node-serialport #1940: Read stream closed

AndrolGenhald, 2019-09-24:
### Summary of Problem

- What are you trying to do?

Read from a serial port which another process may also be trying to read from. This itself is the actual issue of course, but I believe it may have exposed underlying behavior that should still be fixed.

- What happens?

The `serialport` still allows writing data, but will no longer read any data due to this:

```
// zero bytes means read means we've hit EOF? Maybe this should be an error
if (bytesRead === 0) {
  debug('binding.read', 'Zero bytes read closing readable stream')
  this.push(null)
  return
}
```

- What should have happened?

Unless there is a legitimate case where the read stream should be closed due to 0 bytes being read, throwing an error would make it easier to notice issues where something else is reading from the serial port that should not be.

### Code to Reproduce the Issue

Reproduction depends on external factors, namely another process reading from the serial port.

### Versions, Operating System and Hardware

- SerialPort@?

8.0.2

- Node.js v?

12.4.0

- Windows? Linux? Mac?

Linux

- Hardware and chipset? (Prolific/FTDI/Other)

Not really sure...but it's a raspberry pi if that tells you anything.

AndrolGenhald, 2019-09-24:
I also noticed what is probably a bug with unix-read.js:

```
const bytesRead = await readAsync(this.fd, buffer, offset, length, null)
if (bytesRead === 0) {
  return this.read(buffer, offset, length)
}
```

Due to the `promisify`, `readAsync` returns an object like `{bytesRead: integer, buffer: Buffer}`, so `bytesRead === 0` will always be false.

reconbot, 2019-09-25:
That would be a bug.

reconbot, 2019-10-31:
That bug is fixed but this behavior still remains.