serialport/node-serialport #1944: Issue getting device information from docker container

Awoogamuffin, 2019-09-25:
# Summary of Problem

- What are you trying to do?

I'm trying to run .list() from within a docker container

- What happens?

When I run the serialport.list() function on the host machine, I get this information for one of the devices:

```
{ manufacturer: 'Prolific Technology Inc.',
    serialNumber: undefined,
    pnpId: 'usb-Prolific_Technology_Inc._USB-Serial_Controller-if00-port0',
    locationId: undefined,
    vendorId: '067b',
    productId: '2303',
    comName: '/dev/ttyUSB0' }
```

Which is great. But when I run the same from a docker container, I get the following:

```
{ manufacturer: undefined,
     serialNumber: undefined,
     pnpId: undefined,
     locationId: undefined,
     vendorId: undefined,
     productId: undefined,
     comName: '/dev/ttyUSB0' }
```

The container is defined in docker-compose like this:

```
myapp:
   image: myappimage
   ports: 8999:8999
   volumes:
     - /dev:/dev
     - ./:/myappdir
   privileged: true
   command: ./scripts/runApp.sh
```

What's peculiar is that if I remove the `/dev:/dev` from the volumes, and `privileged: true`, the container still accesses all the ports, but with the same result (in that I get not manufacturer or productid information). The various things I've tried for the docker container seem to have almost no effect!

- What should have happened?

The retrieved devices should have all the information within the container as they do without.

### Versions, Operating System and Hardware

Both the host and the Docker container are running ubuntu, to keep things relatively simple.

Awoogamuffin, 2019-10-04:
Issue is solved.

Digging into the linux bindings, I saw that 'udevadm info -e' is used to get all the devices. Running that command within the container gave less information than on the host, but this is what gave me the googleable information I needed to find this page:

https://stackoverflow.com/questions/41753218/udevadm-does-not-show-all-attributes-inside-a-docker-container

So the solution, in my docker-compose file, is this:

```
myapp:
   image: myappimage
   ports: 8999:8999
   volumes:
     - /dev:/dev
     - ./:/myappdir
     - /run/udev:/run/udev:ro
   privileged: true
   command: ./scripts/runApp.sh
```

reconbot, 2019-10-05:
I’m glad to hear that I wasn’t sure how to dig into how docker could expose that info.

closed 2019-10-05 by reconbot