serialport/node-serialport #1945: Prebuild fails for v8.0.2 on Windows with electron-builder

CHH, 2019-09-25:
### Summary of Problem

(Please answer all 3)

- What are you trying to do?

Building an Electron App on Windows, that uses serialport, using electron-builder.

- What happens?

`electron-builder` fails while trying to install prebuilt binaries for serialport bindings v8.0.2 for Electron 6.0.10 (NODE_MODULE_VERSION=73). `prebuild-install` throws a 404, though the package exists on Github:

```
build native dependency from sources  name=@serialport/bindings
                                          version=8.0.2
                                          platform=win32
                                          arch=x64
                                          reason=prebuild-install failed with error (run with env DEBUG=electron-builder to get more information)
                                          error=prebuild-install info begin Prebuild-install version 5.3.2 
    prebuild-install WARN install prebuilt binaries enforced with --force!
    prebuild-install WARN install prebuilt binaries may be out of date!
    prebuild-install info looking for cached prebuild @ C:\Users\ChristophHochstrasse\AppData\Roaming\npm-cache\_prebuilds\0db84a-bindings-v8.0.2-electron-v73-win32-x64.tar.gz
    prebuild-install http request GET https://github.com/serialport/node-serialport/releases/download/v8.0.2/bindings-v8.0.2-electron-v73-win32-x64.tar.gz
    prebuild-install http 404 https://github.com/serialport/node-serialport/releases/download/v8.0.2/bindings-v8.0.2-electron-v73-win32-x64.tar.gz
    prebuild-install WARN install No prebuilt binaries found (target=6.0.1 runtime=electron arch=x64 libc= 
platform=win32)
```
- What should have happened?

Prebuild binaries should be able to download, the archive exists on GitHub.

### Code to Reproduce the Issue

### Versions, Operating System and Hardware

- SerialPort@8.0.2
- Node.js v12 / Electron 6.0.10
- Windows 10 Pro
- Hardware and chipset? Virtual Box VM

Yufeikang, 2019-09-26:
same Problem

anchinoleo, 2019-09-26:
Same problem

reconbot, 2019-09-29:
it's still in beta, working on moving our build infra to github actions. I'm not going to spend the time getting this release built but hopefully the next one will be in better shape.

closed 2019-09-29 by reconbot

bribrah, 2019-10-03:
I have the same problem

patriciussanctus, 2019-10-11:
macos too

JeffJassky, 2019-12-22:
Same. Any idea what's going on here?

JesseWeb, 2019-12-31:
same problem when I building win x64 application on Mac