serialport/node-serialport #1948: Space parity on Mac

toineKeistoer, 2019-09-29:
### Summary of Problem
On Windows everything works good, but when I want to connect the device and run the app on a Mac I cant connect to the device `Binding #open had an error [Error: Invalid parity setting 5]`. 

I read somewhere that parity "space" is not supported on Linux-based systems. I can connect to it using parity "none", "even" and "odd", but I dont receive any messages with those settings. Can I convert the messages somehow from space to none/even/odd parity? 

### Code to Reproduce the Issue

```js
const serialport = new SerialPort(path)
const port = new serialport('/dev/tty.SLAB_USBtoUART', {
    baudRate: 921600,
    dataBits: 8,
    stopBits: 1,
    parity: 'space',
    lock: true,
    autoOpen: true,
    xon: false,
    xoff: false
});
// Code
```

### Versions, Operating System and Hardware
- SerialPort@ ^7.1.5
- Node.js v10.16.0
- Mac
- SLAB_USBtoUART

reconbot, 2019-09-29:
I don’t understand what you’re trying to convert.

toineKeistoer, 2019-09-29:
I don't know converting is possible. I'm looking for a way to connect to the device on a Mac, like I can on Windows.

toineKeistoer, 2019-10-04:
I can connect it with parity "even", but if I do that the messages arent correct.

closed 2019-12-10 by stale[bot]