serialport/node-serialport #1950: Unable to get hardware flow control set up correctly (windows/mac)

gunnarSensus, 2019-10-02:
### Summary of Problem

(Please answer all 3)

- What are you trying to do?
  - i am using node-serialport in an electron project to talk to a hardware device over serial, and my firmware folks have asked if the software can talk over serial using hardware flow control (sending/receiving RTS/CTS signals). things work great without any flow control settings set.
- What happens?
  - i have tried setting the rtscts (also tried other flow control settings, none seem to work) setting when creating my serialport object but i am unable to get consistent communications with the hardware device. my firmware counterpart has been sniffing the wire and can see the hardware sending the rts and cts signals, but the configured software serialport isnt handling those signals and sending when it shouldnt.
- What should have happened?
  - im not an experienced serialport dev, but i believe it should only send when receiving the appropriate signal and listen when receiving the other.
### Code to Reproduce the Issue

```js
    let port = new serialport( path, {'baudRate':  baudRate, rtscts: true});

// Code
```

### Versions, Operating System and Hardware

- SerialPort@7.1.4
- Node.js v8.11.4
- Windows: 10 (various), and 7.  
- Mac: macOS 10.14.6 
- Hardware and chipset: FTDI USB to serial converter chip FT234XD-R.

please let me know if this isnt enough information and i will do my best to get whatever you need.

ErdmanCT13, 2019-10-02:
I too am experiencing this. I've asked one of the engineers at my company about this and apparently we're also using an FTDI chip to handle serial to usb communications. 

Similar to you, we've tried just about everything as far as fiddling with serialport settings. Nothing seems to work. The one glimmer of hope for is was that we've had different results when setting RTS/CTS in Putty. None of the other terminals we tried seemed to work.

ErdmanCT13, 2019-10-02:
> ### Summary of Problem
> (Please answer all 3)
>
> * What are you trying to do?
>
>   * i am using node-serialport in an electron project to talk to a hardware device over serial, and my firmware folks [...]

Try installing SerialPort@8.0.2 from npm. There was a small change made in May that fixed this for us. The previous iteration didn't allow for CTS at all , based on what I saw in current Windows documentation.

gunnarSensus, 2019-10-07:
thanks for the quick response! preliminary results looks like it is working great on windows (waiting on another team member to run some duration tests, but from my personal testing it looks like it is happy the 20 or so times i have tried); which is nice as windows is the only supported platform for the app i am working on, but i am developing on a Mac which isn't working on SerialPort@8.0.2 and it would be nice to be able to develop on my main workstation with flow control enabled. would it be possible to investigate why it isn't happy on Mac? (same versions/OS as before with exception of new serialport version)

closed 2019-12-13 by stale[bot]