serialport/node-serialport #1957: No react-compatible examples

rm-rf-etc, 2019-10-06:
## 📚 Documentation

The recommended solution to build an electron app using react is to run webpack, and webpack "is not supported" by serialport. It seems that there is some known workaround
(https://github.com/serialport/node-serialport/issues/1349#issuecomment-332189381) but what that is, I can't yet tell.

Since react is the most popular framework, I think it makes sense to at least document a method for using serialport with webpack.

rm-rf-etc, 2019-10-08:
The solution I landed on is adding
```
<script>window.serialport = require('serialport')</script>
```
in the head of the `public/index.html` file. Apparently this sidesteps the issues caused by webpack's import system.

reconbot, 2019-10-25:
you want to stick serialport in externals

https://webpack.js.org/configuration/externals/

closed 2019-10-25 by reconbot

fsmiamoto, 2019-11-24:
@rm-rf-etc @reconbot Can you guys help with this problem?
I have created an app with `create-react-app` and added `electron` but still haven't been able to get it working properly.

I've first tried adding the script tag but then I get `require is not defined`

After that, I have used `react-app-rewired` to override the webpack configuration by adding the following:
```js
  config.externals = {
    serialport: {
      commonjs: "serialport",
    }
  };
```
The app doesn't crash anymore but still I get `undefined` when importing `serialport`

If you have any working example that I can take a look, that would help a lot

Thanks!