serialport/node-serialport #1959: yarn add serialport in electron-builder environment got prebuild-install 404 error

CaiBaoHong, 2019-10-24:
yarn add serialport in electron-builder environment got prebuild-install 404 error

My demo project for reproduct the issue is here: https://github.com/CaiBaoHong/serial-port-gui

Reproduct the issue following this step:

```bash
## using vue-cli 3 create a demo project
vue create serial-port-gui
## add electron builder plugin to vue
vue add electron-builder (select version electron 6.0)
## add project dependency: serialport
yarn add serialport
```

when installing the `serialport`, I got this error info:

```bash
caibh@company:~/demo/vue/serial-port-gui$ yarn add serialport
yarn add v1.19.1
[1/4] Resolving packages...
[2/4] Fetching packages...
info fsevents@1.2.9: The platform "linux" is incompatible with this module.
info "fsevents@1.2.9" is an optional dependency and failed compatibility check. Excluding it from installation.
info fsevents@2.1.1: The platform "linux" is incompatible with this module.
info "fsevents@2.1.1" is an optional dependency and failed compatibility check. Excluding it from installation.
[3/4] Linking dependencies...
warning " > less-loader@5.0.0" has unmet peer dependency "webpack@^2.0.0 || ^3.0.0 || ^4.0.0".
[4/4] Building fresh packages...
success Saved lockfile.
success Saved 20 new dependencies.
info Direct dependencies
└─ serialport@7.1.5
info All dependencies
├─ @serialport/bindings@2.0.8
├─ @serialport/parser-byte-length@2.0.2
├─ @serialport/parser-cctalk@2.0.2
├─ @serialport/parser-ready@2.0.2
├─ @serialport/parser-regex@2.0.2
├─ @serialport/stream@2.0.5
├─ bindings@1.5.0
├─ decompress-response@4.2.1
├─ expand-template@2.0.3
├─ file-uri-to-path@1.0.0
├─ github-from-package@0.0.0
├─ napi-build-utils@1.0.1
├─ node-abi@2.12.0
├─ noop-logger@0.1.1
├─ prebuild-install@5.3.2
├─ serialport@7.1.5
├─ simple-concat@1.0.0
├─ simple-get@3.1.0
├─ tar-fs@2.0.0
└─ which-pm-runs@1.0.0
$ electron-builder install-app-deps
  • electron-builder  version=21.2.0
  • rebuilding native dependencies  dependencies=@serialport/bindings@2.0.8 platform=linux arch=x64
  • install prebuilt binary  name=@serialport/bindings version=2.0.8 platform=linux arch=x64
  • build native dependency from sources  name=@serialport/bindings
                                          version=2.0.8
                                          platform=linux
                                          arch=x64
                                          reason=prebuild-install failed with error (run with env DEBUG=electron-builder to get more information)
                                          error=prebuild-install info begin Prebuild-install version 5.3.2
    prebuild-install info looking for cached prebuild @ /home/caibh/.npm/_prebuilds/62a8e4-bindings-v2.0.8-electron-v73-linux-x64.tar.gz
    prebuild-install http request GET https://github.com/node-serialport/node-serialport/releases/download/v2.0.8/bindings-v2.0.8-electron-v73-linux-x64.tar.gz
    prebuild-install http 404 https://github.com/node-serialport/node-serialport/releases/download/v2.0.8/bindings-v2.0.8-electron-v73-linux-x64.tar.gz
    prebuild-install WARN install No prebuilt binaries found (target=6.1.1 runtime=electron arch=x64 libc= platform=linux)
    
  • rebuilding native dependency  name=@serialport/bindings version=2.0.8
Done in 20.84s.
```

I ignore the error of installing `serialport`, and try to test if the api of `serialport` work or not:

*serial-port-gui/src/views/Home.vue:*

```bash
<template>
  <div class="home">
    <button @click="listSerialPorts">list serial ports</button>
  </div>
</template>

<script>
  const serialport = require('serialport')
  export default {
    name: 'home',
    methods: {
      listSerialPorts() {
        serialport.list((err, ports) => {
          ports.forEach(port => console.log('port: %o', port))
        })
      }
    }
  }
</script>
```

when I run `yarn electron:serve`to start up my electron application，got the following error:

```javascript
Uncaught TypeError: Cannot read property 'indexOf' of undefined
    at Function.getFileName (webpack-internal:///./node_modules/bindings/bindings.js:178)
    at bindings (webpack-internal:///./node_modules/bindings/bindings.js:82)
    at eval (webpack-internal:///./node_modules/@serialport/bindings/lib/linux.js:1)
    at Object../node_modules/@serialport/bindings/lib/linux.js (chunk-vendors.js:54)
    at __webpack_require__ (app.js:791)
    at fn (app.js:151)
    at eval (webpack-internal:///./node_modules/@serialport/bindings/lib/index.js:14)
    at Object../node_modules/@serialport/bindings/lib/index.js (chunk-vendors.js:32)
    at __webpack_require__ (app.js:791)
    at fn (app.js:151)
```

Then I find the [issue 1906](https://github.com/serialport/node-serialport/issues/1906) and mark `serialport` as an external by setting your `vue.config.js` to:

```javascript
module.exports = {
  pluginOptions: {
    electronBuilder: {
      externals: ['serialport']
    }
  }
}
```

Run `yarn electron:serve` again, the `serialport` api works !

But when I run `yarn`, the problem `prebuild-install http 404 https://github.com/node-serialport/node-serialport/releases/download/v2.0.8/bindings-v2.0.8-electron-v73-linux-x64.tar.gz` is still there, I just want to known why?

wtilton, 2019-10-25:
Pretty much getting the same error. Added serial port to the externals and still nothing. npm install also doesn't want to work. :(

CaiBaoHong, 2019-10-25:
> Pretty much getting the same error. Added serial port to the externals and still nothing. npm install also doesn't want to work. :(

have you try to use `yarn` instead `npm`?

CaiBaoHong, 2019-10-25:
I know what is the error info mean now ! Just see the following error info again:
```bash
$ electron-builder install-app-deps
  • electron-builder  version=21.2.0
  • rebuilding native dependencies  dependencies=@serialport/bindings@2.0.8 platform=linux arch=x64
  • install prebuilt binary  name=@serialport/bindings version=2.0.8 platform=linux arch=x64
  • build native dependency from sources  name=@serialport/bindings
                                          version=2.0.8
                                          platform=linux
                                          arch=x64
                                          reason=prebuild-install failed with error (run with env DEBUG=electron-builder to get more information)
                                          error=prebuild-install info begin Prebuild-install version 5.3.2
    prebuild-install info looking for cached prebuild @ /home/caibh/.npm/_prebuilds/62a8e4-bindings-v2.0.8-electron-v73-linux-x64.tar.gz
    prebuild-install http request GET https://github.com/node-serialport/node-serialport/releases/download/v2.0.8/bindings-v2.0.8-electron-v73-linux-x64.tar.gz
    prebuild-install http 404 https://github.com/node-serialport/node-serialport/releases/download/v2.0.8/bindings-v2.0.8-electron-v73-linux-x64.tar.gz
    prebuild-install WARN install No prebuilt binaries found (target=6.1.1 runtime=electron arch=x64 libc= platform=linux)
    
  • rebuilding native dependency  name=@serialport/bindings version=2.0.8
```
It means yarn try to build native dependency of `@serialport/binding version 2.0.8` with the following step :
1. try to find cache in local
2. try to download from github repo if exist, actually not found, so it got response 404
3. if both not found in cache or github, build by myself

ps:
I ty to build in windows system, it need:
1. install python2.7
2. `npm install --global --production windows-build-tools`, remember to open power shell with Administrator
3. `yarn add serialport`

closed 2019-10-25 by CaiBaoHong

wtilton, 2019-10-25:
> > Pretty much getting the same error. Added serial port to the externals and still nothing. npm install also doesn't want to work. :(
> 
> have you try to use `yarn` instead `npm`?

Yes, been trying yarn as well.

wtilton, 2019-10-25:
Ran through the other steps you mentioned, using yarn this time. Plus I updated to electron 7. This might be more of an electron issue maybe now.
```
An unhandled error occurred inside electron-rebuild
Could not detect abi for version 7.0.0 and runtime electron.  Updating "node-abi" might help solve this issue if it is a new release of electron
```

wtilton, 2019-10-25:
This is what I'm getting after doing an npm install.

```
> electron-rebuild

× Rebuild Failed

An unhandled error occurred inside electron-rebuild
gyp WARN install got an error, rolling back install
gyp ERR! configure error
gyp ERR! stack Error: 403 status code downloading arm64 node.lib
gyp ERR! stack     at Request.<anonymous> (c:\projects\focuscar\CanBusSniffa\node_modules\electron-rebuild\node_modules\node-gyp\lib\install.js:335:22)
gyp ERR! stack     at Request.emit (events.js:194:15)
gyp ERR! stack     at Request.onRequestResponse (c:\projects\focuscar\CanBusSniffa\node_modules\request\request.js:1066:10)
gyp ERR! stack     at ClientRequest.emit (events.js:189:13)
gyp ERR! stack     at HTTPParser.parserOnIncomingClient [as onIncoming] (_http_client.js:556:21)
gyp ERR! stack     at HTTPParser.parserOnHeadersComplete (_http_common.js:109:17)
gyp ERR! stack     at TLSSocket.socketOnData (_http_client.js:442:20)
gyp ERR! stack     at TLSSocket.emit (events.js:189:13)
gyp ERR! stack     at addChunk (_stream_readable.js:284:12)
gyp ERR! stack     at readableAddChunk (_stream_readable.js:265:11)
gyp ERR! System Windows_NT 10.0.18362
gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "c:\\projects\\focuscar\\CanBusSniffa\\node_modules\\electron-rebuild\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild" "--target=6.0.0" "--arch=x64" "--dist-url=https://electronjs.org/headers" "--build-from-source"
gyp ERR! cwd c:\projects\focuscar\CanBusSniffa\node_modules\@serialport\bindings
gyp ERR! node -v v10.15.3
gyp ERR! node-gyp -v v5.0.5
gyp ERR! not ok

Failed with exit code: 1

Error: gyp WARN install got an error, rolling back install
gyp ERR! configure error
gyp ERR! stack Error: 403 status code downloading arm64 node.lib
gyp ERR! stack     at Request.<anonymous> (c:\projects\focuscar\CanBusSniffa\node_modules\electron-rebuild\node_modules\node-gyp\lib\install.js:335:22)
gyp ERR! stack     at Request.emit (events.js:194:15)
gyp ERR! stack     at Request.onRequestResponse (c:\projects\focuscar\CanBusSniffa\node_modules\request\request.js:1066:10)
gyp ERR! stack     at ClientRequest.emit (events.js:189:13)
gyp ERR! stack     at HTTPParser.parserOnIncomingClient [as onIncoming] (_http_client.js:556:21)
gyp ERR! stack     at HTTPParser.parserOnHeadersComplete (_http_common.js:109:17)
gyp ERR! stack     at TLSSocket.socketOnData (_http_client.js:442:20)
gyp ERR! stack     at TLSSocket.emit (events.js:189:13)
gyp ERR! stack     at addChunk (_stream_readable.js:284:12)
gyp ERR! stack     at readableAddChunk (_stream_readable.js:265:11)
gyp ERR! System Windows_NT 10.0.18362
gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "c:\\projects\\focuscar\\CanBusSniffa\\node_modules\\electron-rebuild\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild" "--target=6.0.0" "--arch=x64" "--dist-url=https://electronjs.org/headers" "--build-from-source"
gyp ERR! cwd c:\projects\focuscar\CanBusSniffa\node_modules\@serialport\bindings
gyp ERR! node -v v10.15.3
gyp ERR! node-gyp -v v5.0.5
gyp ERR! not ok

Failed with exit code: 1
    at SafeSubscriber._error (c:\projects\focuscar\CanBusSniffa\node_modules\spawn-rx\lib\src\index.js:267:84)
    at SafeSubscriber.__tryOrUnsub (c:\projects\focuscar\CanBusSniffa\node_modules\rxjs\internal\Subscriber.js:205:16)
    at SafeSubscriber.error (c:\projects\focuscar\CanBusSniffa\node_modules\rxjs\internal\Subscriber.js:156:26)
    at Subscriber._error (c:\projects\focuscar\CanBusSniffa\node_modules\rxjs\internal\Subscriber.js:92:26)
    at Subscriber.error (c:\projects\focuscar\CanBusSniffa\node_modules\rxjs\internal\Subscriber.js:72:18)
    at MapSubscriber.Subscriber._error (c:\projects\focuscar\CanBusSniffa\node_modules\rxjs\internal\Subscriber.js:92:26)
    at MapSubscriber.Subscriber.error (c:\projects\focuscar\CanBusSniffa\node_modules\rxjs\internal\Subscriber.js:72:18)
    at SafeSubscriber._next (c:\projects\focuscar\CanBusSniffa\node_modules\spawn-rx\lib\src\index.js:242:65)
    at SafeSubscriber.__tryOrUnsub (c:\projects\focuscar\CanBusSniffa\node_modules\rxjs\internal\Subscriber.js:205:16)
    at SafeSubscriber.next (c:\projects\focuscar\CanBusSniffa\node_modules\rxjs\internal\Subscriber.js:143:22)
```

wtilton, 2019-10-25:
> I know what is the error info mean now ! Just see the following error info again:

Anyway you could share your working code? I just want to get this working and actually move on to making an app.

wtilton, 2019-10-25:
Again not sure, totally new to all of this but there's this issue over in the node-gyp code https://github.com/nodejs/node-gyp/issues/1933

CaiBaoHong, 2019-10-25:
> Ran through the other steps you mentioned, using yarn this time. Plus I updated to electron 7. This might be more of an electron issue maybe now.
>
> ```
> An unhandled error occurred inside [...]

I suggestyou use electron version 6.0, not version 7.0, is too new

wtilton, 2019-10-25:
> I suggestyou use electron version 6.0, not version 7.0, is too new

Yeah just reverted, thought the same thing, getting that error above. Is this where you got your info to do the installs?

https://github.com/nodejs/node-gyp

I went through that, maybe I need a restart. I'll try that now.

wtilton, 2019-10-25:
Restarting didn't work. Seems like something's wrong with node-gyp.

After a yarn
```
An unhandled error occurred inside electron-rebuild
gyp info it worked if it ends with ok
gyp info using node-gyp@5.0.5
gyp info using node@10.15.3 | win32 | x64
gyp info find Python using Python version 2.7.15 found at "C:\Users\rball\.windows-build-tools\python27\python.exe"
gyp http GET https://electronjs.org/headers/v6.0.0/node-v6.0.0-headers.tar.gz
gyp http 200 https://electronjs.org/headers/v6.0.0/node-v6.0.0-headers.tar.gz
gyp http GET https://electronjs.org/headers/v6.0.0/SHASUMS256.txt
gyp http GET https://electronjs.org/headers/v6.0.0/win-x86/node.lib
gyp http GET https://electronjs.org/headers/v6.0.0/win-x64/node.lib
gyp http GET https://electronjs.org/headers/v6.0.0/win-arm64/node.lib
gyp http 200 https://electronjs.org/headers/v6.0.0/win-x64/node.lib
gyp http 403 https://electronjs.org/headers/v6.0.0/win-arm64/node.lib
gyp WARN install got an error, rolling back install
gyp http 200 https://electronjs.org/headers/v6.0.0/SHASUMS256.txt
gyp http 200 https://electronjs.org/headers/v6.0.0/win-x86/node.lib
gyp ERR! configure error
gyp ERR! stack Error: 403 status code downloading arm64 node.lib
gyp ERR! stack     at Request.<anonymous> (C:\projects\focuscar\CanBusSniffa\node_modules\electron-rebuild\node_modules\node-gyp\lib\install.js:335:22)
gyp ERR! stack     at Request.emit (events.js:194:15)
gyp ERR! stack     at Request.onRequestResponse (C:\projects\focuscar\CanBusSniffa\node_modules\request\request.js:1066:10)
gyp ERR! stack     at ClientRequest.emit (events.js:189:13)
gyp ERR! stack     at HTTPParser.parserOnIncomingClient [as onIncoming] (_http_client.js:556:21)
gyp ERR! stack     at HTTPParser.parserOnHeadersComplete (_http_common.js:109:17)
gyp ERR! stack     at TLSSocket.socketOnData (_http_client.js:442:20)
gyp ERR! stack     at TLSSocket.emit (events.js:189:13)
gyp ERR! stack     at addChunk (_stream_readable.js:284:12)
gyp ERR! stack     at readableAddChunk (_stream_readable.js:265:11)
gyp ERR! System Windows_NT 10.0.18362
gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\projects\\focuscar\\CanBusSniffa\\node_modules\\electron-rebuild\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild" "--target=6.0.0" "--arch=x64" "--dist-url=https://electronjs.org/headers" "--build-from-source"
gyp ERR! cwd C:\projects\focuscar\CanBusSniffa\node_modules\@serialport\bindings
gyp ERR! node -v v10.15.3
gyp ERR! node-gyp -v v5.0.5
gyp ERR! not ok

Failed with exit code: 1

Error: gyp info it worked if it ends with ok
gyp info using node-gyp@5.0.5
gyp info using node@10.15.3 | win32 | x64
gyp info find Python using Python version 2.7.15 found at "C:\Users\rball\.windows-build-tools\python27\python.exe"
gyp http GET https://electronjs.org/headers/v6.0.0/node-v6.0.0-headers.tar.gz
gyp http 200 https://electronjs.org/headers/v6.0.0/node-v6.0.0-headers.tar.gz
gyp http GET https://electronjs.org/headers/v6.0.0/SHASUMS256.txt
gyp http GET https://electronjs.org/headers/v6.0.0/win-x86/node.lib
gyp http GET https://electronjs.org/headers/v6.0.0/win-x64/node.lib
gyp http GET https://electronjs.org/headers/v6.0.0/win-arm64/node.lib
gyp http 200 https://electronjs.org/headers/v6.0.0/win-x64/node.lib
gyp http 403 https://electronjs.org/headers/v6.0.0/win-arm64/node.lib
gyp WARN install got an error, rolling back install
gyp http 200 https://electronjs.org/headers/v6.0.0/SHASUMS256.txt
gyp http 200 https://electronjs.org/headers/v6.0.0/win-x86/node.lib
gyp ERR! configure error
gyp ERR! stack Error: 403 status code downloading arm64 node.lib
gyp ERR! stack     at Request.<anonymous> (C:\projects\focuscar\CanBusSniffa\node_modules\electron-rebuild\node_modules\node-gyp\lib\install.js:335:22)
gyp ERR! stack     at Request.emit (events.js:194:15)
gyp ERR! stack     at Request.onRequestResponse (C:\projects\focuscar\CanBusSniffa\node_modules\request\request.js:1066:10)
gyp ERR! stack     at ClientRequest.emit (events.js:189:13)
gyp ERR! stack     at HTTPParser.parserOnIncomingClient [as onIncoming] (_http_client.js:556:21)
gyp ERR! stack     at HTTPParser.parserOnHeadersComplete (_http_common.js:109:17)
gyp ERR! stack     at TLSSocket.socketOnData (_http_client.js:442:20)
gyp ERR! stack     at TLSSocket.emit (events.js:189:13)
gyp ERR! stack     at addChunk (_stream_readable.js:284:12)
gyp ERR! stack     at readableAddChunk (_stream_readable.js:265:11)
gyp ERR! System Windows_NT 10.0.18362
gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\projects\\focuscar\\CanBusSniffa\\node_modules\\electron-rebuild\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild" "--target=6.0.0" "--arch=x64" "--dist-url=https://electronjs.org/headers" "--build-from-source"
gyp ERR! cwd C:\projects\focuscar\CanBusSniffa\node_modules\@serialport\bindings
gyp ERR! node -v v10.15.3
gyp ERR! node-gyp -v v5.0.5
gyp ERR! not ok

Failed with exit code: 1
    at SafeSubscriber._error (C:\projects\focuscar\CanBusSniffa\node_modules\spawn-rx\lib\src\index.js:267:84)
    at SafeSubscriber.__tryOrUnsub (C:\projects\focuscar\CanBusSniffa\node_modules\rxjs\internal\Subscriber.js:205:16)
    at SafeSubscriber.error (C:\projects\focuscar\CanBusSniffa\node_modules\rxjs\internal\Subscriber.js:156:26)
    at Subscriber._error (C:\projects\focuscar\CanBusSniffa\node_modules\rxjs\internal\Subscriber.js:92:26)
    at Subscriber.error (C:\projects\focuscar\CanBusSniffa\node_modules\rxjs\internal\Subscriber.js:72:18)
[... cut ...] \rxjs\internal\Subscriber.js:92:26)
    at MapSubscriber.Subscriber.error (C:\projects\focuscar\CanBusSniffa\node_modules\rxjs\internal\Subscriber.js:72:18)
    at SafeSubscriber._next (C:\projects\focuscar\CanBusSniffa\node_modules\spawn-rx\lib\src\index.js:242:65)
    at SafeSubscriber.__tryOrUnsub (C:\projects\focuscar\CanBusSniffa\node_modules\rxjs\internal\Subscriber.js:205:16)
    at SafeSubscriber.next (C:\projects\focuscar\CanBusSniffa\node_modules\rxjs\internal\Subscriber.js:143:22)
```

wtilton, 2019-10-25:
Seems to be an issue with a lot of people https://spin.atomicobject.com/2019/03/27/node-gyp-windows/

CaiBaoHong, 2019-10-25:
> > I know what is the error info mean now ! Just see the following error info again:
> 
> Anyway you could share your working code? I just want to get this working and actually move on to making an app.

My demo has updated and add document in README.md, hope to help you. https://github.com/CaiBaoHong/serial-port-gui

wtilton, 2019-10-25:
> My demo has updated and add document in README.md, hope to help you. https://github.com/CaiBaoHong/serial-port-gui

I'll try going through your demo steps. Maybe that'll work. I can't get node-gyp to work by itself so fingers crossed.

wtilton, 2019-10-25:
On step 2.2.2 I get the following. I have windows-build-tools and Python27 installed. I did the default windows-build-tools so maybe it did 2019? Maybe I need 2017? I did install with the flag 2015

```
$ electron-builder install-app-deps
  • electron-builder  version=21.2.0
  • rebuilding native dependencies  dependencies=@serialport/bindings@2.0.8 platform=win32 arch=x64
  • install prebuilt binary  name=@serialport/bindings version=2.0.8 platform=win32 arch=x64
  • build native dependency from sources  name=@serialport/bindings
                                          version=2.0.8
                                          platform=win32
                                          arch=x64
                                          reason=prebuild-install failed with error (run with env DEBUG=electron-builder to get more information)
                                          error=prebuild-install info begin Prebuild-install version 5.3.2
    prebuild-install info looking for cached prebuild @ C:\Users\rball\AppData\Roaming\npm-cache\_prebuilds\a3ea41-bindings-v2.0.8-electron-v73-win32-x64.tar.gz
    prebuild-install http request GET https://github.com/node-serialport/node-serialport/releases/download/v2.0.8/bindings-v2.0.8-electron-v73-win32-x64.tar.gz
    prebuild-install http 404 https://github.com/node-serialport/node-serialport/releases/download/v2.0.8/bindings-v2.0.8-electron-v73-win32-x64.tar.gz
    prebuild-install WARN install No prebuilt binaries found (target=6.1.2 runtime=electron arch=x64 libc= platform=win32)

  • rebuilding native dependency  name=@serialport/bindings version=2.0.8
Done in 16.77s.
```

wtilton, 2019-10-25:
MS has a visual installer so I'm going to go through that and just install all of them (again).

CaiBaoHong, 2019-10-26:
you don't have to care the warn info in step 2.2.2, because the work of install serialport lib is done

wtilton, 2019-10-28:
No idea what's up, but I tried it on a different machine and serialport works. On the other machine when I bring up the app and getting:

```
Uncaught Error: Could not locate the bindings file. Tried:
 → c:\projects\focuscar\CanBusSniffa\node_modules\@serialport\bindings\build\bindings.node
 → c:\projects\focuscar\CanBusSniffa\node_modules\@serialport\bindings\build\Debug\bindings.node
 → c:\projects\focuscar\CanBusSniffa\node_modules\@serialport\bindings\build\Release\bindings.node
 → c:\projects\focuscar\CanBusSniffa\node_modules\@serialport\bindings\out\Debug\bindings.node
 → c:\projects\focuscar\CanBusSniffa\node_modules\@serialport\bindings\Debug\bindings.node
 → c:\projects\focuscar\CanBusSniffa\node_modules\@serialport\bindings\out\Release\bindings.node
 → c:\projects\focuscar\CanBusSniffa\node_modules\@serialport\bindings\Release\bindings.node
 → c:\projects\focuscar\CanBusSniffa\node_modules\@serialport\bindings\build\default\bindings.node
 → c:\projects\focuscar\CanBusSniffa\node_modules\@serialport\bindings\compiled\12.4.0\win32\x64\bindings.node
 → c:\projects\focuscar\CanBusSniffa\node_modules\@serialport\bindings\addon-build\release\install-root\bindings.node
 → c:\projects\focuscar\CanBusSniffa\node_modules\@serialport\bindings\addon-build\debug\install-root\bindings.node
 → c:\projects\focuscar\CanBusSniffa\node_modules\@serialport\bindings\addon-build\default\install-root\bindings.node
```

I think this is happening because npm install doesn't work on this machine, but it doesn't work on the other one either. Either way, the bindings folder is completely missing on this machine. I tried copying over the files and got a version error. So maybe this is also due to different versions of node, gyp or something else. Going to install the older version of node on here and see if that works.