serialport/node-serialport #1960: delimiter loss data

tookbra, 2019-10-26:
### Summary of Problem

- What are you trying to do?
used delimiter to split data by Frame header is 0x5A

- What happens?
when I run the program, Console output
```
<Buffer 00 01 02 10 00 01 00 29 b5>
<Buffer 00 01 12 00 00 0d 00 06 20 19 03 20 00 70 18 00 01 01 44 ab 01>
```
but when I used Device Monitoring Studio to monitoring serial port
[image]
```
5A 00 01 02 10 00 01 00 29 B5
5A 00 01 12 00 00 0D 00 06 20 19 03 20 00 70 18 00 01 01 44 AB 01
5A 00 01 12 01 00 01 00 40 FC
```

where is ‘5A 00 01 12 01 00 01 00 40 FC’， the data listener did not return

- What should have happened?

### Code to Reproduce the Issue

```js
var SerialPort = require('serialport');
var Delimiter = SerialPort.parsers.Delimiter;
var parser = serialPort.pipe(new Delimiter({delimiter: [90]}));

var serialPort = new SerialPort('COM3', {
  baudRate: 115200,
  autoOpen: false
});

serialPort.open(function(error) {
  console.log('isOpen:', serialPort.isOpen);
  if (error) {
    console.log('open port failed：' + error);
  } else {
    let cmd = Buffer.from([90, 0, 1, 2, 16, 0, 5, 0, 0, 0, 15, 0, -57, -87]);
    serialPort.write(cmd3, function (error) {
     if (error) console.log(‘error’);
     else {
       console.log(new Date());
      }
    });
  }
});

serialPort.on('data', function (data) {
  console.log(data);
});
```

### Versions, Operating System and Hardware

- SerialPort@?7.1.5
- Node.js v?v10.16.3
- Windows
- Hardware and chipset? (Prolific/FTDI/Other)

reconbot, 2019-10-26:
Delimiter can’t know how long your packet will be and without a next header
it won’t know the packet ended.

tookbra, 2019-10-27:
tks, i got it.

closed 2019-10-27 by tookbra