serialport/node-serialport #1961: Error: Writing to COM port (GetOverlappedResult): Unknown error code 31

toineKeistoer, 2019-10-26:
### Summary of Problem
Since a few weeks I'm often (not always) getting an error when trying to write messages. The errors **Writing to COM port (GetOverlappedResult): Unknown error code 31** and **Reading from COM port (ReadIOCompletion): Operation aborted** occures and the connection is closed. Sometimes, randomly, when I unplug the device and power it on/off it works again (but only the first try).

Does anyone know this issue?

```
 serialport/bindings loading WindowsBinding +0ms
  serialport/stream opening path: COM3 +0ms
  serialport/binding-abstract open +0ms
  serialport/stream _read queueing _read for after open +8ms
  serialport/stream opened path: COM3 +2s
  serialport/stream _read reading +12ms
  serialport/binding-abstract read +2s
  serialport/stream _write 1 bytes of data +21ms
  serialport/binding-abstract write 1 bytes +5ms
  serialport/stream binding.write error [Error: Writing to COM port (GetOverlappedResult): Unknown error code 31] +32ms
  serialport/stream disconnected [Error: Writing to COM port (GetOverlappedResult): Unknown error code 31] +5ms
  serialport/stream #close +2ms
  serialport/binding-abstract close +37ms
> Error Writing to COM port (GetOverlappedResult): Unknown error code 31
  serialport/stream binding.read error { [Error: Reading from COM port (ReadIOCompletion): Operation aborted] canceled: true } +43ms
  serialport/stream _read queueing _read for after open +4ms
  serialport/stream binding.close finished +2ms
> Close
```

### Code to Reproduce the Issue

```js
const SerialPort = require('serialport');
const messages = {
    logoff: Buffer.from([0x86, 0x00, 0x86]),
    logon: Buffer.from([0x86, 0x01, 0x87]),
    logall: Buffer.from([
        0x85,
        0x00,
        0x08,
        0x00,
        0x00,
        0x00,
        0x00,
        0x1f,
        0xff,
        0xff,
        0xff,
        0x00,
        0x00,
        0x00,
        0x00,
        0xa9
    ])
};

const port = new SerialPort('COM3', {
    baudRate: 921600,
    dataBits: 8,
    stopBits: 1,
    parity: 'space',
    lock: true,
    autoOpen: false,
    xon: false,
    xoff: false
});

port.on('close', (line) => console.log(`> Close`));
port.on('error', (err) => console.log(`> Error`, err.message));
port.on('data', (data) => console.log(data.toString('HEX')));
port.open(() => {
    port.write('V');
    port.write(messages.logall);
    port.write(messages.logon);
});
```

### Versions, Operating System and Hardware

- SerialPort@6.11.3
- Node.js v10.16.3
- Windows Server 2016
- Hardware and chipset: 
  comName: 'COM3',
  manufacturer: 'Silicon Labs',
  serialNumber: '04660ed6d432e911bc9f4545fa452d63',
  pnpId: 'USB\\VID_10C4&PID_EA60\\04660ED6D432E911BC9F4545FA452D63',
  locationId: 'Port_#0006.Hub_#0003',
  vendorId: '10C4',
  productId: 'EA60'

closed 2020-01-02 by stale[bot]

fvilante, 2022-01-07:
I'm having same issue. Asking for `never-stale` label

fvilante, 2022-01-07:
I solved the same problem. 
In my case I was using a Laptop and the usb was shutting down (bad connection).

ShanArosh, 2023-03-17:
Now I'm having the same could anyone please help me