serialport/node-serialport #1975: Angular + Electron + serialport | Issues with Typescript

ThomasEcherer, 2019-10-31:
### Summary of Problem

- What are you trying to do?
i played around with serialport. Got it working.
Expanded with Electron -> Got it working.
Expanded further with an Angular app -> Can not get it working :-(

- What happens?
As soon as i try to interact with serialport by typescript the program crashes

- What should have happened?
A Typescript Variation would be nice

### Code to Reproduce the Issue

i installed also "@types/serialport": "^7.0.4",

```ts
// none of the following is working
const SerialPort = require('serialport'); 
import * as SerialPort from 'serialport';

...

async readAvailablePorts() {
    const ports = await SerialPort.list();
    console.log('Available SerialPorts: ', ports);
  }
```

### Versions, Operating System and Hardware

- Angular: 8.2.13
- Electron: 7.0.0
- SerialPort@? 8.0.5
- Node.js v? 12.13.0
- Windows? Linux? Mac? Windows 10
- Hardware and chipset? (Prolific/FTDI/Other) FTDI

reconbot, 2019-11-01:
What's the error message?

ThomasEcherer, 2019-11-04:
yeah, sorry. here some additional informations:

i installed following packages:

```
$ npm install electron --save
$ npm install  electron-rebuild --save-dev

$ npm install serialport --save
$ npm install @types/serialport --save-dev

$ npm install @types/node --save-dev
```

i ran : 
`$ ./node_modules/.bin/electron-rebuild`
to rebuild it for my system

i tried to start the application with:
`ng build --prod && electron .`

That causes an error. The error messages looks like this:

> Field 'browser' doesn't contain a valid alias configuration
>     resolve as module
>       looking for modules in C:/dev/test/myProject/
>         using description file: C:\dev\test\myProject\package.json (relative path: .)
>           Field 'browser' doesn't contain a valid alias configuration
>           using description file: C:\dev\test\myProject\package.json (relative path: ./stream)
>             no extension
>               Field 'browser' doesn't contain a valid alias configuration
>               C:\dev\test\myProject\stream doesn't exist
>             .ts
>               Field 'browser' doesn't contain a valid alias configuration
>               C:\dev\test\myProject\stream.ts doesn't exist
>             .tsx
>               Field 'browser' doesn't contain a valid alias configuration
>               C:\dev\test\myProject\stream.tsx doesn't exist
>             .mjs
>               Field 'browser' doesn't contain a valid alias configuration
>               C:\dev\test\myProject\stream.mjs doesn't exist
>             .js
>               Field 'browser' doesn't contain a valid alias configuration
>               C:\dev\test\myProject\stream.js doesn't exist
>             as directory
>               C:\dev\test\myProject\stream doesn't exist

ThomasEcherer, 2019-11-04:
ok i got it working now.

i've added a custom webpack to load serialport by that way.

by installing:
```
npm i @angular-builders/custom-webpack
npm i @angular-builders/dev-server
npm i @angular-devkit/build-angular
```

changes in angular.json:
```
  "architect": {
    "build": {
      "builder": "@angular-builders/custom-webpack:browser",
      "options": {
        "customWebpackConfig": {"path": "./custom-webpack.config.js"},
...
"serve": {
      "builder": "@angular-builders/dev-server:generic",
```

added cutom-webpack.config.js file:
```
module.exports = {
  "externals": {
    "serialport": "require('serialport')"
  }
}
```

now the build is successfull. 
i ran into an other problem with "Uncaught ReferenceError: require is not defined".
But that is not caused by serialport, so i close this ticket

closed 2019-11-04 by ThomasEcherer