serialport/node-serialport #1977: UnhandledPromise Warning, SerialPort.list, no callback, return promise...

epikao, 2019-10-31:
Hello, I'm new at JS, and Node, but below Code which I have from the official documentation side (V5.0...) does not work, and gives allways follow error:
"UnhandledPromise...Warning:.... SerialPort.list no longer takes a callback.... only return a promise".

Unfortunately I have no idea, why this code should not work :-( ?

```
var SerialPort = require('serialport');
// callback approach
SerialPort.list(function (err, ports) {
  ports.forEach(function(port) {
    console.log(port.comName);
    console.log(port.pnpId);
    console.log(port.manufacturer);
  });
});
```

thanks

reconbot, 2019-11-01:
If you're using serialport 8 as the error says "SerialPort.list no longer takes a callback and only returns a promise". You're providing a callback and not using the returned promise. Additionally `comName` has been renamed `path`. See the [upgrade guide](https://github.com/serialport/node-serialport/blob/master/UPGRADE_GUIDE.md) and the [list() docs](https://serialport.io/docs/api-stream#serialportlist) for other changes and examples.

You want to do;
```
const SerialPort = require('serialport');
// Promise approach
SerialPort.list().then(ports => {
  ports.forEach(function(port) {
    console.log(port.path);
    console.log(port.pnpId);
    console.log(port.manufacturer);
  });
});
```

Where did you find the example you're using?

epikao, 2019-11-02:
thank you, that works now :-)
the example I found several time in the internet, for example stackoverflow etc...

reconbot, 2019-11-09:
great!

closed 2019-11-09 by reconbot

sagar-vandure, 2020-01-23:
great, thank you its help full to me to solve same error

halilaykent, 2020-02-06:
2/6/2020, 21:23:56] Homebridge is running
g on port 51826.
Unhandled Rejection TypeError: Serialport
.list no longer takes a callback and only
returns a promise
at Function.SerialPort.list (/usr/loc
al/lib/node modules/homebridge-zigbee-v2/
node modules/@serialport/stream/lib/index
•js:651:11)
at /usr/local/lib/node_modules/homebrew
idge-zigbee-v2/lib/utils/findSerialPort.j

s:5:16

at new Promise ()
at find Serial Port (/usr/local/lib/node
e modules/homebridge-zigbee-v2/lib/utils/
findSerialPort.js:4:10),
at ZigBee Platform.start ZigBee (/usr/l
ocal/lib/node_modules/homebridge-zigbee-v
2/index.js:69:39)
at ZigBee Platform.handle Initialization
n (/usr/local/lib/node_modules/homebridge
-zigbee-v2/index.js:64:10)
at API.emit (events.js:223:5),
at Server.run(/usr/local/lib/node_mo
dules/homebridge/lib/server.js:103:13)
at module.exports (/usr/local/lib/node
node_modules/homebridge/lib/cli.js:59:10),
at Object. (/usr/local/lib
/node_modules/homebridge/bin/homebridge:1
7:22)