serialport/node-serialport #1979: Initialize SerialPort instance without specifying path to port

richardstefun, 2019-11-02:
### Summary of Problem

(Please answer all 3)

- What are you trying to do?
I'm trying to implement custom module for Vue.js that is imported in Vuex (store) and is called using external functions
- What happens?
I am not able to initialize 'blank' SerialPort instance. I'm new to JS, so there may be something I'm missing. I'm doing this because that path may change so I get it's name from store (that also retrieves it from JSON saved in device storage). I created SerialPort instance with path called 'placeholder' and then renamed it in port.open with a **correct** path name from my function that is called using store, but I only get error TypeError: callback.call is not a function at SerialPort._error after opening. I understand that this is because SerialPort class does not have any setters for path name that would allow me to change it later.
- What should have happened?
I would like to create blank SerialPort instance and then set path to it later, and open it with specific method later
### Code to Reproduce the Issue

```js
import store from './store'
const SerialPort = require('serialport')
const port = new SerialPort('placeholder', { baudRate: 38400, autoOpen: false });
const Delimiter = require('@serialport/parser-delimiter')
const parser = port.pipe(new Delimiter({ delimiter: '\n' }))
parser.on('data', console.log)

port.on('open', function() {
  // open logic
})

export const reconnect = () => {
    if (port.isOpen) {
        port.close();
    }
    let portName = store.getters.currentPort;
    
    console.log(portName);
    port.open(portName, {
        baudRate: 38400
    }, function (err) {
        console.log('Error: ', err.message)
    });
}

```

### Versions, Operating System and Hardware
**- Electron "^6.0.12"**
- SerialPort@^7.1.5
- Node.js v12.12.0
- Windows? Linux? Mac? Mac OS 10.15
- Hardware and chipset? (Prolific/FTDI/Other) CH340

reconbot, 2019-11-03:
This isn't a compatible pattern. The serialport object needs to be created in a working state. You can use a wrapper object that creates the serialport and closes it as needed if you need this patten for vue.

epikao, 2019-11-08:
> This isn't a compatible pattern. The serialport object needs to be created in a working state. You can use a wrapper object that creates the serialport and closes it as needed if you need this patten for vue.

this is also exactly the same I'm trying to do since several days now, but without success... the problem is that I don't really understand JS with all his options :-( ... I would very appreciate some code examples...

reconbot, 2019-11-10:
I've never really used angular but something like this might work? (untested)

```js
class PortController {
  constructor() {
    this.path = null
    this.port = null
  }  
  get isOpen() {
    if (!this.port) {
      return false
    }
    return this.port.isOpen
  }
  open(path, cb) {
    if(this.isOpen) {
      throw new Error('Port already open')
    }
    this.path = path
    this.port = new SerialPort(path, cb)
  }
  close(cb) {
    if (this.port) {
      this.port.close(cb)
    }
  }
}```

You'll probably also want to handle when the port is opening (between calling `open()` and when `port.isOpen` turns true) and closing but that's up to you and your application.

closed 2019-11-10 by reconbot