serialport/node-serialport #1980: callback if serialport is disconnected

epikao, 2019-11-02:
Hello, I try to understand how to get a callback if serialport is disconnected, also I have studied issue #764, but unfortunately I don't understand this solution.

https://github.com/serialport/node-serialport/pull/764/commits/f4ba90c46a209e62749e864535f4135d5a391872

follow code does not work (windows 10):
```
port.on('disconnect', function(err){
    console.log("disconnected");
}); 
```
thank you very much

epikao, 2019-11-03:
or differently asked, how to disconnect the serialport if already opened?

A disconnection I can recognize with a little work around, with the serialport.list.

reconbot, 2019-11-03:
You’ll get a close event with an error object.

epikao, 2019-11-03:
Hello, thanks for your reply.

if I unplug my serial hardware, I don't get any callback/error with code see below.
Additional, how can I disconnect the serialport in software?
serialport.close(); does not work.

```
serialport.on('error', function(err) {
  console.log('Error: ', err.message)
});
```

reconbot, 2019-11-03:
You can close a port with the close method.
- https://serialport.io/docs/api-stream#serialport-close

If a disconnection happens you'll get a close event with an error.
- https://serialport.io/docs/api-stream#close

epikao, 2019-11-04:
thank you, regarding the disconnect/ unplug hardware issue, I still not get any callback / event:

```
serialport.on('close', function(err){
console.log("Port closed.");
});

serialport.on('disconnect', function(err){
console.log("Port closed.");
  if(err.disconnected === true){
    console.log("disconnected");}
}); 
```

reconbot, 2019-11-04:
There is no disconnect event. Unless you are reading or writing we don’t know the port was discontent.

epikao, 2019-11-05:
ok thanks.
another question, I know is a little bit off-topic, but maybe even though you can give me a hint:

```
...
..
serialport.close();
nextfunction();
..
```
I want start the "nextfunction();" after serialport.close() is finished and not before. Unfortunately my JS knowledge is to bad to find a solution... I already have tried with Async/Await - but no success :-(...

thanks

reconbot, 2019-11-06:
You can use callbacks or wrap it in a promise.

```
serialport.close((err) => {
  if(err) {/* do something */}
  nextFunction()
})
```
or 
```
async function() {
  await new Promise((resolve, reject) => serialport.close(err => err ? reject(err): resolve()))
  nextFunction()
}
```

epikao, 2019-11-06:
ok. but unfortunately that still not works as expected, see below
```
function xy()
{
     for(var i=0; i <= 2; i++)
     {
     console.log("first");
     serialport.close((err) => { if(err){console.log("closed");}  });
     console.log("second");
    }
}
```

**console result is as follow:**
first
second
first 
second
closed
closed

**but I expect:**
first
closed
second
first
closed
second
```

reconbot, 2019-11-06:
Close is an async action and for is a sync loop. The close callback won’t
fire until later.

epikao, 2019-11-06:
> Close is an async action and for is a sync loop. The close callback won’t
> fire until lat

ok. seems a little bit more complicated, at least with my current knowledge :-/

epikao, 2019-11-07:
can you help me again? see follow code below, the serialport.close(); gives me always "error port is not open" instead to close the port (the serialport.open() function works fine):

```
function startserial(serialcom,openclose,idx)
{
var	serialport = new SerialPort(serialcom,{
    baudRate: 115200,
    databits: 8,
    parity: 'even',
    stopbits: 1,
    flowControl: false,
	autoOpen: false
	});

	if(openclose=="open")
	{
		console.log("serialport.open", serialcom);
		serialport.open();}
		
	if(openclose=="close")
	{
		console.log("serialport.close", serialcom);
		serialport.close();}

serialport.on('open', function(err) {

	console.log('Port opened');
	comdata[idx]=""+comdata[idx]+" open";

});

serialport.on('error', function(err) {
  console.log('Error: ', err.message);
 
  arr=comdata[idx].split(" ");
  comdata[idx]=""+arr[0]+" "+arr[1];

});

serialport.on('close', function(err) {
	console.log("Port closed.");
	
	arr=comdata[idx].split(" ");
	comdata[idx]=""+arr[0]+" "+arr[1];	
});

}
```

closed 2020-01-15 by stale[bot]