serialport/node-serialport #1981: WorkerThreads: Cannot create a handle without a HandleScope

meisterlampe, 2019-11-08:
### Summary of Problem
- What are you trying to do?
I want to open a serial connection from within a WorkerThread.
Maybe I'm using this wrong, any hint how to do it right is much appreciated :-)

- What happens?
I get this Error:
```
FATAL ERROR: HandleScope::HandleScope Entering the V8 API without proper locking in place
 1: 0x9d8da0 node::Abort() [node]
 2: 0x9d9f56 node::OnFatalError(char const*, char const*) [node]
 3: 0xb37b8a v8::Utils::ReportApiFailure(char const*, char const*) [node]
 4: 0xb3910e v8::HandleScope::HandleScope(v8::Isolate*) [node]
 5: 0x9498ae node::InternalCallbackScope::InternalCallbackScope(node::Environment*, v8::Local<v8::Object>, node::async_context const&, node::InternalCallbackScope::ResourceExpectation) [node]
 6: 0x94a7c5 node::MakeCallback(v8::Isolate*, v8::Local<v8::Object>, v8::Local<v8::Function>, int, v8::Local<v8::Value>*, node::async_context) [node]
 7: 0x7fd0a4c6c042 EIO_AfterOpen(uv_work_s*) [/home/dev/git/test/serialport-vs-workerthread/node_modules/@serialport/bindings/build/Release/bindings.node]
 8: 0x12cfdd5  [node]
 9: 0x12d42d1  [node]
10: 0x12e66a8  [node]
11: 0x12d4c5b uv_run [node]
12: 0xa1afa7 node::NodeMainInstance::Run() [node]
13: 0x9ab298 node::Start(int, char**) [node]
14: 0x7fd0a7a9db97 __libc_start_main [/lib/x86_64-linux-gnu/libc.so.6]
15: 0x949035  [node]
Aborted (core dumped)
```
- What should have happened?
SerialPort should work as normal.

### Code to Reproduce the Issue

Minimal reproduction, that leads to the error:
```js
const { Worker } = require('worker_threads');

const worker = new Worker(
    `new require("serialport")("/dev/ttyUSB0");`,
     { eval: true }
);
```

Running the same new outside the worker works fine:
```js
const { Worker } = require('worker_threads');
new require("serialport")("/dev/ttyUSB0");
```

Also requiring another library (tried moment here) works as expected:
```js
const { Worker } = require('worker_threads');

const worker = new Worker(
    `console.log((new require("moment"))().toISOString());`,
     { eval: true }
);
```

### Versions, Operating System and Hardware

- SerialPort@8.0.5
- Node.js v12.13.0
- Linux (Xubuntu 18.04)

Could this be related to some child-thread serialport is trying to start?
Please help :-)

reconbot, 2019-11-09:
We don’t yet support running in workers.

meisterlampe, 2019-11-09:
Thank you for your fast answer. 
Is support possible/planned?
Any hints what has to be implemented to make it work?

reconbot, 2019-11-09:
Quite a bit of c++ work has to be done to make it happen. Basically we need
to port to NAPI. A few people have started to work on it but we haven’t
landed any work towards that end.

meisterlampe, 2019-11-09:
OK. Thanks for the good answers and of course for this lib too. I guess I can close this question. :)

closed 2019-11-09 by meisterlampe

reconbot, 2019-11-09:
Actually looking at the previous tickets here, we need some c++ hooks and
maybe it’s an easier change than porting from NAN to N-API.

meisterlampe, 2019-11-11:
Does the addon need to be "context aware" as described in here https://nodejs.org/api/addons.html#addons_context_aware_addons ?

reconbot, 2019-11-11:
It’s the cleanup hooks

https://nodejs.org/api/addons.html#addons_worker_support

meisterlampe, 2019-11-11:
Really looks easier than porting from nan to n-api (at least the description is shorter :-D ). 
Maybe I should continue commenting on Issue #1938