serialport/node-serialport #1984: Unable to import on docker

anacierdem, 2019-11-09:
### Summary of Problem

(Please answer all 3)

- What are you trying to do?
import the library inside docker container
- What happens?
throws;

```/workspaces/ed64/node_modules/bindings/bindings.js:91
        throw e
        ^

Error: /workspaces/ed64/node_modules/@serialport/bindings/build/Release/bindings.node: invalid ELF header
    at Object.Module._extensions..node (module.js:681:18)
    at Module.load (module.js:565:32)
    at tryModuleLoad (module.js:505:12)
    at Function.Module._load (module.js:497:3)
    at Module.require (module.js:596:17)
    at require (internal/module.js:11:18)
    at bindings (/workspaces/ed64/node_modules/bindings/bindings.js:84:48)
    at Object.<anonymous> (/workspaces/ed64/node_modules/@serialport/bindings/lib/linux.js:1:98)
    at Module._compile (module.js:652:30)
    at Object.Module._extensions..js (module.js:663:10)```
- What should have happened?

### Code to Reproduce the Issue

```js
const SerialPort = require('serialport')

// Code
```

### Versions, Operating System and Hardware

- SerialPort@? 7.1.0
- Node.js v? v8.10.0
- Windows? Linux? Mac? Ubuntu 18.04 on docker
- Hardware and chipset? (Prolific/FTDI/Other) N/A

reconbot, 2019-11-09:
It should download the right binaries during install, if your code has a `node_modules` that was built on a different system than it can't run the binaries it downloaded. If you're installing `node_modules` during `docker build` please provide that output. You might be able to add a `npm rebuild` in your dockerfile to get it to just recompile whatever you're providing.

anacierdem, 2019-11-10:
Ok, I'm pretty sure that's it thank you!

closed 2019-11-10 by anacierdem