serialport/node-serialport #1985: open and close serialport with "path"

epikao, 2019-11-09:
### Summary of Problem

(Please answer all 3)

- What are you trying to do?
open and close port, with fix given path as a string "COM7" it works, but with a variable "serialcom" for path, it does not work.
- What happens?
with a variable to path serialport it does not work, only if path is a fix value.
- What should have happened?
open and close alternating

### Code to Reproduce the Issue

```js
const serialport = new SerialPort(path)

const SerialPort = require('serialport');

var serialcom = " ";
var i = 0;
	
//if path is "COM7" it works, but not if path is a variable like "serialcom"?!?
var	serialport = new SerialPort(serialcom,{
    baudRate: 115200,
    databits: 8,
    parity: 'even',
    stopbits: 1,
    flowControl: false,
	autoOpen: false
	});

serialport.on("open", function(err) {

	console.log('Port opened');

});

serialport.on("error", function(err) {
  console.log('Error: ', err.message);

});

serialport.on("close", function() {
	console.log("Port closed.");

});

		setInterval(function() {
			
			
			i++;
			if(i==1)
			{
				console.log("open");
				serialcom="COM7"; //that does not work, only if "COM7" is directly fixed in the Serialport Path.
				serialport.open();
				
			}
			if(i==2)
			{
				console.log("close");
				serialcom="COM7"; //that does not work, only if "COM7" is directly fixed in the Serialport Path.
				serialport.close();
				i=0;
			}
				

		},
		2000);
```

### Versions, Operating System and Hardware

- SerialPort@?
- Node.js v?
- **Windows**
- FTDI

epikao, 2019-11-10:
I have simplified my question above, hope somebody now can help me... 
I have lost now several days to solve this issue but without success :-(

reconbot, 2019-11-11:
You need to make a new serialport object each time.

epikao, 2019-11-11:
> You need to make a new serialport object each time.

I have tried this already, but if do so it never calls function "showPortOpen", and the other ones.
See my code below:

```
const SerialPort = require('serialport');

var serialcom = " ";
var i = 0;

var	port = new SerialPort(serialcom,{
    baudRate: 115200,
    databits: 8,
    parity: 'even',
    stopbits: 1,
    flowControl: false,
	autoOpen: false
	});

port.on('open', showPortOpen);
port.on('data', readSerialData);
port.on('close', showPortClose);
port.on('error', showError);
	

function showPortOpen() {
   console.log('port open. Data rate: ' + port.baudRate);
}
 
function readSerialData(data) {
   console.log(data);
}
 
function showPortClose() {
   console.log('port closed.');
}
 
function showError(error) {
   console.log('Serial port error: ' + error);
}

setInterval(function() {						
		i++;
		if(i==1)
		{
			console.log("open");
			serialcom="COM3";
			port = new SerialPort(serialcom);		
			//serialport.open();
	
		}
		if(i==2)
		{
			console.log("close");
			//serialcom="COM3";
			port.close();
			i=0;
		}				
},
2000);
```

closed 2019-11-11 by HipsterBrown