serialport/node-serialport #1992: Binding version mismatch (compiled at 72, expecting 75) in Electron

iamcam, 2019-11-15:
I've searched through the related issues and solutions, but none seem to fix my issue. Running as a node process from the CLI, everything is fine, however when running within electron, I get a NODE_MODULE_VERSION mismatch.

When running `electron .`, I get the error below:

```
Uncaught Exception:
Error: The module '/Users/Cameron/Development/{...}/node_modules/@serialport/bindings/build/Release/bindings.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 72. This version of Node.js requires
NODE_MODULE_VERSION 75. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).
    at process.func (electron/js2c/asar.js:140:31)
    at process.func [as dlopen] (electron/js2c/asar.js:140:31)
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:922:18)
    at Object.func (electron/js2c/asar.js:140:31)
    at Object.func [as .node] (electron/js2c/asar.js:140:31)
    at Module.load (internal/modules/cjs/loader.js:735:32)
    at Module._load (internal/modules/cjs/loader.js:648:12)
    at Module._load (electron/js2c/asar.js:717:26)
    at Function.Module._load (electron/js2c/asar.js:717:26)
    at Module.require (internal/modules/cjs/loader.js:775:19)
```

# Current Versions

* **MacOS**: Mojave 10.14.6
* **serialport**:  8.0.5
* **Node (via nvm)** : 12.8.1, (also LTS/erbium)
* **python**: 2.7.10
* **Electron**:

```
{
  node: '12.8.1',
  v8: '7.8.279.23-electron.0',
  uv: '1.30.1',
  zlib: '1.2.11',
  brotli: '1.0.7',
  ares: '1.15.0',
  modules: '75',
  nghttp2: '1.39.2',
  napi: '4',
  llhttp: '1.1.4',
  http_parser: '2.8.0',
  openssl: '1.1.0',
  icu: '64.2',
  unicode: '12.1',
  electron: '7.1.1',
  chrome: '78.0.3904.99'
}
```

# What I've tried:
* Generally - the attempts in this open issue - #1910 
* `nvm i` + `nvm rebuild` :  no fix
* `electron-rebuild`: Failed. This fails with several repeated C++ errors (see bottom of this report)
* I've tried both matching and mismatching versions of node and electron with no difference
* deleting all node modules and starting fresh: no change
* [These steps](https://github.com/serialport/node-serialport/issues/1910#issuecomment-524949720): no fix 
* Running node-gyp directly in the bindings directory reproduces the same C++ errors listed below.

# Observations

* No matter what I try, electron attempts to load the v72 bindings, _even though electron v75_ bindings are available in the GitHub release.

Any help would be greatly appreciated. 

# `electron-rebuild` errors

```
 Rebuild Failed

An unhandled error occurred inside electron-rebuild
  CXX(target) Release/obj.target/bindings/src/serialport.o
  CXX(target) Release/obj.target/bindings/src/serialport_unix.o
  CXX(target) Release/obj.target/bindings/src/poller.o
  CXX(target) Release/obj.target/bindings/src/darwin_list.o
In file included from ../src/darwin_list.cpp:3:
In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/IOKit.framework/Headers/IOKitLib.h:45:
In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/CoreFoundation.framework/Headers/CFBase.h:72:
/usr/local/include/Block.h:16:3: error: Never include this file directly. Use <lzma.h> instead.
#       error Never include this file directly. Use <lzma.h> instead.
        ^
/usr/local/include/Block.h:93:2: error: unknown type name 'lzma_check'
        lzma_check check;
        ^
/usr/local/include/Block.h:148:2: error: unknown type name 'lzma_vli'
        lzma_vli compressed_size;
        ^
```

...etc...

```
make: *** [Release/obj.target/bindings/src/darwin_list.o] Error 1
gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/Users/cameron/Development/{...}/node_modules/node-gyp/lib/build.js:194:23)
gyp ERR! stack     at ChildProcess.emit (events.js:203:13)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:272:12)
gyp ERR! System Darwin 18.7.0
gyp ERR! command "/Users/cameron/.nvm/versions/node/v12.8.1/bin/node" "/Users/cameron/Development/{...}/node_modules/.bin/node-gyp" "rebuild" "--target=7.1.1" "--arch=x64" "--dist-url=https://electronjs.org/headers" "--build-from-source"
gyp ERR! cwd /Users/cameron/Development/{...}/node_modules/@serialport/bindings
gyp ERR! node -v v12.8.1
gyp ERR! node-gyp -v v5.0.5
gyp ERR! not ok

Failed with exit code: 1

Error:   CXX(target) Release/obj.target/bindings/src/serialport.o
  CXX(target) Release/obj.target/bindings/src/serialport_unix.o
  CXX(target) Release/obj.target/bindings/src/poller.o
  CXX(target) Release/obj.target/bindings/src/darwin_list.o
In file included from ../src/darwin_list.cpp:3:
In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/IOKit.framework/Headers/IOKitLib.h:45:
In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/CoreFoundation.framework/Headers/CFBase.h:72:
/usr/local/include/Block.h:16:3: error: Never include this file directly. Use <lzma.h> instead.
#       error Never include this file directly. Use <lzma.h> instead.
        ^
/usr/local/include/Block.h:93:2: error: unknown type name 'lzma_check'
        lzma_check check;
```

...etc...

iamcam, 2019-11-15:
Followup... I did some additional searching regarding the CXX target errors and came up with this similar thread + solution for a Golang issue: https://github.com/golang/go/issues/25240#issuecomment-458224865

Once I remove (or in my case tgz and remove the original), I'm able to run `electron-rebuild` successfully. The electron app now works as expected.

It seems that the rebuild steps do work for some people, and maybe the general instructions can be added to the documentation. What I don't know is whether or not this is because I have a lot of random things installed on my machine from years of dev work, or if this is more widespread.

o1xhack, 2019-11-27:
@iamcam  Can you tell me what you removed for? I've got a similar problem with another lib and want to try out your way

iamcam, 2019-11-27:
There was an issue with the particular copy of block.h (maybe from some old lib I had installed before the package managers started using usr/local). By removing the one at /usr/local/include/block.h everything cleared up and the build process could use the right version.  Just in case I tar’d up my copy instead of outright deleting.

closed 2019-11-27 by iamcam