serialport/node-serialport #1993: Electron rebuild fails at compiling serialport

eMarek, 2019-11-19:
### Summary of Problem

I am trying to rebuild serialport with electron-rebuild, but it fails with bunch of errors:
```
my-computer:my-app my-username$ electron-rebuild
✖ Rebuild Failed

An unhandled error occurred inside electron-rebuild
gyp info it worked if it ends with ok
gyp info using node-gyp@6.0.1
gyp info using node@12.13.1 | darwin | x64
gyp info find Python using Python version 3.7.4 found at "/usr/local/opt/python/bin/python3.7"
gyp info spawn /usr/local/opt/python/bin/python3.7
gyp info spawn args [
gyp info spawn args   '/Users/my-username/Documents/bdi/my-app/node_modules/node-gyp/gyp/gyp_main.py',
gyp info spawn args   'binding.gyp',
gyp info spawn args   '-f',
gyp info spawn args   'make',
gyp info spawn args   '-I',
gyp info spawn args   '/Users/my-username/Documents/bdi/my-app/node_modules/firmata/node_modules/@serialport/bindings/build/config.gypi',
gyp info spawn args   '-I',
gyp info spawn args   '/Users/my-username/Documents/bdi/my-app/node_modules/node-gyp/addon.gypi',
gyp info spawn args   '-I',
gyp info spawn args   '/Users/my-username/.electron-gyp/7.1.1/include/node/common.gypi',
gyp info spawn args   '-Dlibrary=shared_library',
gyp info spawn args   '-Dvisibility=default',
gyp info spawn args   '-Dnode_root_dir=/Users/my-username/.electron-gyp/7.1.1',
gyp info spawn args   '-Dnode_gyp_dir=/Users/my-username/Documents/bdi/my-app/node_modules/node-gyp',
gyp info spawn args   '-Dnode_lib_file=/Users/my-username/.electron-gyp/7.1.1/<(target_arch)/node.lib',
gyp info spawn args   '-Dmodule_root_dir=/Users/my-username/Documents/bdi/my-app/node_modules/firmata/node_modules/@serialport/bindings',
gyp info spawn args   '-Dnode_engine=v8',
gyp info spawn args   '--depth=.',
gyp info spawn args   '--no-parallel',
gyp info spawn args   '--generator-output',
gyp info spawn args   'build',
gyp info spawn args   '-Goutput_dir=.'
gyp info spawn args ]
gyp info spawn make
gyp info spawn args [ 'BUILDTYPE=Release', '-C', 'build' ]
  CXX(target) Release/obj.target/bindings/src/serialport.o
../src/serialport.cpp:329:14: error: no matching member function for call to 'Set'
    results->Set(Nan::New<v8::String>("cts").ToLocalChecked(), Nan::New<v8::Boolean>(data->cts));
    ~~~~~~~~~^~~
/Users/my-username/.electron-gyp/7.1.1/include/node/v8.h:3455:37: note: candidate function not viable: requires 3 arguments, but 2 were provided
  V8_WARN_UNUSED_RESULT Maybe<bool> Set(Local<Context> context,
                                    ^
/Users/my-username/.electron-gyp/7.1.1/include/node/v8.h:3458:37: note: candidate function not viable: requires 3 arguments, but 2 were provided
  V8_WARN_UNUSED_RESULT Maybe<bool> Set(Local<Context> context, uint32_t index,
                                    ^
../src/serialport.cpp:330:14: error: no matching member function for call to 'Set'
    results->Set(Nan::New<v8::String>("dsr").ToLocalChecked(), Nan::New<v8::Boolean>(data->dsr));
    ~~~~~~~~~^~~
/Users/my-username/.electron-gyp/7.1.1/include/node/v8.h:3455:37: note: candidate function not viable: requires 3 arguments, but 2 were provided
  V8_WARN_UNUSED_RESULT Maybe<bool> Set(Local<Context> context,
                                    ^
/Users/my-username/.electron-gyp/7.1.1/include/node/v8.h:3458:37: note: candidate function not viable: requires 3 arguments, but 2 were provided
  V8_WARN_UNUSED_RESULT Maybe<bool> Set(Local<Context> context, uint32_t index,
                                    ^
../src/serialport.cpp:331:14: error: no matching member function for call to 'Set'
    results->Set(Nan::New<v8::String>("dcd").ToLocalChecked(), Nan::New<v8::Boolean>(data->dcd));
    ~~~~~~~~~^~~
/Users/my-username/.electron-gyp/7.1.1/include/node/v8.h:3455:37: note: candidate function not viable: requires 3 arguments, but 2 were provided
  V8_WARN_UNUSED_RESULT Maybe<bool> Set(Local<Context> context,
                                    ^
/Users/my-username/.electron-gyp/7.1.1/include/node/v8.h:3458:37: note: candidate function not viable: requires 3 arguments, but 2 were provided
  V8_WARN_UNUSED_RESULT Maybe<bool> Set(Local<Context> context, uint32_t index,
                                    ^
../src/serialport.cpp:378:14: error: no matching member function for call to 'Set'
    results->Set(Nan::New<v8::String>("baudRate").ToLocalChecked(), Nan::New<v8::Integer>(data->baudRate));
    ~~~~~~~~~^~~
[... 127 lines cut ...]
  },
  "devDependencies": {
    "cross-env": "^6.0.3",
    "electron": "^7.1.1",
    "electron-rebuild": "^1.8.8",
    "node-gyp": "^6.0.1"
  },
  "dependencies": {
    "johnny-five": "^1.3.1",
    "serialport": "^8.0.5"
  }
}
```

### Versions, Operating System and Hardware

- SerialPort@8.0.5
- Node.js v12.13.1
- npm 6.12.1
- node-gyp v6.0.1
- yarn 1.19.1
- macOS Catalina Version 10.15.1
- Python 2.7.16

I already opened same issue at [electron-rebuild](https://github.com/electron/electron-rebuild/issues/320), but they directed me here. I am a bit lost now.

closed 2019-11-30 by eMarek

Raylin51, 2020-03-30:
Do you fix this problem? I would like to know how to fix it. Thanks!

eMarek, 2020-03-30:
In my case problem was in outdated `serialport` inside `johnny-five` package. It was fixed by updating related packages. Take a look at https://github.com/electron/electron-rebuild/issues/320 and https://github.com/rwaldron/johnny-five/issues/1643.

Raylin51, 2020-03-31:
Got it. Thanks!