serialport/node-serialport #1994: prebuild-install gets 404 when downloading windows bindings (but not Mac)

iamcam, 2019-11-20:
When attempting to build electron with electron-build for windows via the docker image, there's an issue with getting a 404 when attempting to load the windows bindings from GitHub. The Mac version builds just fine, as does linux from the Docker image.

In Docker, If I manually download the bindings from the releases the copy the file name it missed (eg the has-prefixed filename) into _prebuilds folder, it builds fine. It appears to use the prebuilt binary on the Mac side as well.

```
    prebuild-install WARN install prebuilt binaries enforced with --force!
    prebuild-install WARN install prebuilt binaries may be out of date!
    prebuild-install info looking for cached prebuild @ /Users/XXXXXX/.npm/_prebuilds/ac1814-bindings-v8.0.4-electron-v75-win32-x64.tar.gz
    prebuild-install http request GET https://github.com/serialport/node-serialport/releases/download/v8.0.4/bindings-v8.0.4-electron-v75-win32-x64.tar.gz
    prebuild-install http 404 https://github.com/serialport/node-serialport/releases/download/v8.0.4/bindings-v8.0.4-electron-v75-win32-x64.tar.gz
    prebuild-install WARN install No prebuilt binaries found (target=7.1.1 runtime=electron arch=x64 libc= platform=win32)
```

Is there something missing in the prerebuild configuration where its looking for the wrong URL or something? 

Any help would be greatly appreciated.

therealtbs, 2019-11-21:
Looking at the URLs that are requested, prebuild-install seems to be looking for a release called `v8.0.4`, however the release in this repo with the artifacts is called `@serialport/bindings@8.0.4`. 

I'm not sure if this is something that can be easily fixed on the side of the serialport team, however I used the following manual workaround:

1. Look at the logged output. The following information is important:
      1. the line 'looking for cached prebuild @' note the path following the @-sign
      2. the line 'request GET' note the filename at the end of the URL.
2. Download the file from the correct URL to the location from step 1. I used curl 
```bash
curl -L "https://github.com/serialport/node-serialport/releases/download/%40serialport%2Fbindings%408.0.4/$FILENAME" -o "$PATH_TO_CACHED"
```

JeffJassky, 2019-12-22:
This is still happening. Anybody know what the issue is?

reconbot, 2019-12-25:
However `electron-builder` 

```json
{
  "scripts": {
    "install": "prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild",
  }
}
```

I just landed #2003 (thanks @s00d!) which gives electron-builder a way to pickup the tag prefix by default, unknown why it's not being passed through on windows, I'd open an issue with them.

reconbot, 2019-12-25:
I published `@serialport/bindings@8.0.6` the binaries are building now, please try it out!

JesseWeb, 2019-12-27:
it still get 404 when I building Windows app on Mac using electron-builder.

s00d, 2019-12-27:
> it still get 404 when I building Windows app on Mac using electron-builder.

Best way it use a docker container electronuserland/builder:wine for build windows and Linux
[multi-platform-build](https://www.electron.build/multi-platform-build)

JesseWeb, 2019-12-31:
> > it still get 404 when I building Windows app on Mac using electron-builder.
> 
> Best way it use a docker container electronuserland/builder:wine for build windows and Linux
> [multi-platform-build](https://www.electron.build/multi-platform-build)
 
I am testing electronuserland/builder:wine for building win64 application ,still get this error. 
```
 electron-builder  version=22.2.0 os=4.9.184-linuxkit
  • loaded configuration  file=package.json ("build" field)
  • electron-rebuild not required if you use electron-builder, please consider to remove excess dependency from devDependencies

To ensure your native dependencies are always matched electron version, simply add script `"postinstall": "electron-builder install-app-deps" to your `package.json`
  • writing effective config  file=build/builder-effective-config.yaml
  • executing node-gyp rebuild  arch=x64
  • rebuilding native dependencies  dependencies=@serialport/bindings@8.0.6 platform=win32 arch=x64
  • install prebuilt binary  name=@serialport/bindings version=8.0.6 platform=win32 arch=x64
  ⨯ cannot build native dependency  reason=prebuild-install failed with error and build from sources not possible because platform or arch not compatible
                                    cause=exit status 1
                                    errorOut=prebuild-install info begin Prebuild-install version 5.3.3
    prebuild-install info looking for cached prebuild @ /root/.npm/_prebuilds/64b73e-bindings-v8.0.6-electron-v75-win32-x64.tar.gz
    prebuild-install http request GET https://github.com/serialport/node-serialport/releases/download/v8.0.6/bindings-v8.0.6-electron-v75-win32-x64.tar.gz
    prebuild-install http 404 https://github.com/serialport/node-serialport/releases/download/v8.0.6/bindings-v8.0.6-electron-v75-win32-x64.tar.gz
    prebuild-install WARN install No prebuilt binaries found (target=7.1.7 runtime=electron arch=x64 libc= platform=win32)
    
                                    command=/usr/local/bin/node /project/node_modules/prebuild-install/bin.js --platform=win32 --arch=x64 --target=7.1.7 --runtime=electron --verbose --force
                                    workingDir=/project/node_modules/@serialport/bindings
  ⨯ /project/node_modules/app-builder-lib/node_modules/app-builder-bin/linux/x64/app-builder exited with code ERR_ELECTRON_BUILDER_CANNOT_EXECUTE  stackTrace=
                                                                                                                                                     Error: /project/node_modules/app-builder-lib/node_modules/app-builder-bin/linux/x64/app-builder exited with code ERR_ELECTRON_BUILDER_CANNOT_EXECUTE
                                                                                                                                                         at ChildProcess.childProcess.once.code (/project/node_modules/app-builder-lib/node_modules/builder-util/src/util.ts:239:14)
                                                                                                                                                         at Object.onceWrapper (events.js:285:13)
                                                                                                                                                         at ChildProcess.emit (events.js:197:13)
                                                                                                                                                         at maybeClose (internal/child_process.js:984:16)
                                                                                                                                                         at Process.ChildProcess._handle.onexit (internal/child_process.js:265:5)
error Command failed with exit code 1.
```

s00d, 2020-01-05:
Sorry its my mistake, I used the format of the old version. I make new Pull request [2008](https://github.com/serialport/node-serialport/pull/2008)

reconbot, 2020-01-05:
Where can I find the docs on that?

s00d, 2020-01-06:
[here](https://github.com/dominictarr/rc#standards)

typester, 2020-01-23:
I confirmed that the fix in #2008 resolved this issue, but there is no released version that is included this fix.
I hope it will be released soon.

jsgv, 2020-01-29:
Any reason why there has not been release for this?

reconbot, 2020-01-30:
No reason but my day job keeping me busy.

Successfully published:
 - serialport@8.0.7
 - @serialport/bindings@8.0.7
 - @serialport/list@8.0.7
 - @serialport/repl@8.0.7
 - @serialport/terminal@8.0.7

let me know how it works

mettz, 2020-02-15:
Hi, I run into the same issue because version 8 of Electron uses NODE_MODULE_VERSION 76 and most recent serialport release is for version 75.
Here's the output from electron-builder
```prebuild-install info looking for cached prebuild @ /Users/mattia/.npm/_prebuilds/ad07bf-bindings-v8.0.7-electron-v76-linux-x64.tar.gz
prebuild-install http request GET https://github.com/serialport/node-serialport/releases/download/@serialport/bindings@8.0.7/bindings-v8.0.7-electron-v76-linux-x64.tar.gz
prebuild-install http 404 https://github.com/serialport/node-serialport/releases/download/@serialport/bindings@8.0.7/bindings-v8.0.7-electron-v76-linux-x64.tar.gz
prebuild-install WARN install No prebuilt binaries found (target=8.0.0 runtime=electron arch=x64 libc= platform=linux)
```
To make it work I downgraded my app to version 7.1.12 of electron but I made this post to informs core developers of the problem.

mkurczewski, 2020-02-17:
> Hi, I run into the same issue because version 8 of Electron uses NODE_MODULE_VERSION 76 and most recent serialport release is for version 75.
> Here's the output from electron-builder
> 
> ```
> prebuild-install http request GET https://github.com/serialport/node-serialport/releases/download/@serialport/bindings@8.0.7/bindings-v8.0.7-electron-v76-linux-x64.tar.gz
> prebuild-install http 404 https://github.com/serialport/node-serialport/releases/download/@serialport/bindings@8.0.7/bindings-v8.0.7-electron-v76-linux-x64.tar.gz
> prebuild-install WARN install No prebuilt binaries found (target=8.0.0 runtime=electron arch=x64 libc= platform=linux)
> ```
> 
> To make it work I downgraded my app to version 7.1.12 of electron but I made this post to informs core developers of the problem.

This is the issue I reported few days ago https://github.com/serialport/node-serialport/issues/2031

mkurczewski, 2020-02-28:
Hi @reconbot, what's the approximate date when we can see the update for Electron 8?

reconbot, 2020-03-02:
What needs to be done to support it?

mkurczewski, 2020-03-06:
@reconbot As per https://github.com/serialport/node-serialport/issues/2031 we just need a bindings for `electron-v76`

ghost602, 2020-04-20:
I set it back from version 8 to 7.
I am using it now.

reconbot, 2020-05-10:
`serialport@9` now ships with ABI v76 for electron

closed 2020-07-18 by stale[bot]

bbb169, 2024-01-25:
I found another way to solve this
I found a fork of robotjs, and I use this library to replace robotjs, https://github.com/hurdlegroup/robotjs